chipsalliance/rocket-chip #2877: BundleMap: Connect clients with different user field to one crossbar

ljwljwljwljw, 2021-09-12:
**Type of issue**: other enhancement

**Impact**: API modification

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

I got the following error when connecting two clients (one with XXXField in user bits, one without) to a TLXbar:
```
requirement failed: Attempt to assign BundleMap(Wire in TLXbar) :<= BundleMap(Wire in TLXbar), where RHS is missing directional field XXXField()
```

**What is the current behavior?**

TLXbar seems to require all clients to have the same user field.

**What is the expected behavior?**

Would it be better if TLXbar could automatically fill in missing fields with their default values?

**What is the use case for changing the behavior?**

For large systems, it may be difficult to require all clients on TLXbar to have the same user field.

mwachs5, 2021-09-12:
Isn't that what `UserYanker`s are for? Tile link attempts to separate concerns of each widget. Stripping user bits is a different task than arbitrating requests.

ljwljwljwljw, 2021-09-22:
Sorry for the late reply, I found that the reason for the error was that I used BundleMap incorrectly, for example:
```
case class XXXField() extends BundleField(XXXKey) {
  override def data: Bool = Bool()
  override def default(x: Bool): Unit = true.B
}
```
The correct usage should be:
```
case class XXXField() extends BundleField(XXXKey) {
  override def data: Bool = Output(Bool())
  override def default(x: Bool): Unit = { x := true.B }
}
```
After the above correction, the missing user field will be automatically set to its default value when connected to TLXbar.

In addition, 
>Isn't that what UserYankers are for? The TileLink implementation in this repo attempts to separate concerns of each widget. Stripping user bits is a different task than arbitrating requests.

there seems to be no TLUserYanker in the current tilelink package, only AXI4UserYanker in amba.axi4?