chipsalliance/rocket-chip #2884: [chisel-3.5] new chisel3.cover clashes with freechips.rocketchip.util.property.cover

ekiwi, 2021-09-27:
Just a heads up that the next release of Chisel is going to have its own `chisel3.cover(Bool, String)` and that is going to clash with `freechips.rocketchip.util.property.cover` in quite some places.
There are two solutions here:
1) change the uses of `cover` to explicitly use `property.cover`
2) I would be happy to work with you all to try and make the `chisel3.cover` a good drop in replacement for your old custom solution

Kevin

ekiwi, 2021-09-27:
@debs-sifive

debs-sifive, 2021-09-27:
Thanks for the heads up! We should be ok with explicitly stating which cover to use for now.

michael-etzkorn, 2022-05-10:
Handled here: https://github.com/chipsalliance/rocket-chip/commit/92077511baedf2c8a07f7774d5c272682a975b4b

closed 2022-05-10 by michael-etzkorn