chipsalliance/rocket-chip #2887: RoCC Memory interface get stucked 2.0

Phantom1003, 2021-09-29:
I meet the same problem as described in #1753 .
My original design will send the request valid signal ignoring the response valid signal.
So I take the advice in #1753 , waiting for the previous response valid signal.
But my design still waits for the request ready signal, which keeps 0 to the end.
[image]
The red line in the picture shows that all requests wait for previous responses.
I guess maybe is the tag conflict of the cache request, mu rocc current request is 0x42(orange), while 0x42 also appears in the processor requests.
I wonder if this is the true reason cause my design stall?
Which tag range should be used by rocc?

charlekinghy, 2022-07-04:
Hello, @Phantom1003 . I find that you mentioned that your design has large contiguous memory access requests. I am working on finding a way for data exchange between my coprocessor and the core, too. I found that io.cmd and io.resp only support 64bit data tranfer. Can you share the method that support the data exchange of a large chunk? Is there a way to exchange data between the coprocessor and the dcache?

charlekinghy, 2022-07-05:
Thank you for your reply. Do you mean that the length of data exchange between the core and the coprocessor as well as the dcache is limited to 64bit when the core is a normal RISC-V core?