chipsalliance/rocket-chip #2893: Confusion on rocketcore writeback logic?

jerryhethatday, 2021-10-16:
Say we have two  instructions, the first is a load operation that is missed( so, according to the code base, dcache module wont  assert `io.dmem.resp.valid` high when the load operation is at its wb stage), the second one is an operation which will write registers duing its write back stage(wxd). Is there chances that the first load operation that is missed assert   `io.dmem.resp.valid` high when the second operation is at write back stage.  Under this circumanstance, seems that the load result of the first instruction will be write to regfile at this cycle:
```    
val rf_wdata = Mux(dmem_resp_valid && dmem_resp_xpu, io.dmem.resp.bits.data(xLen-1, 0),
                 Mux(ll_wen, ll_wdata,
                 Mux(wb_ctrl.csr =/= CSR.N, csr.io.rw.rdata,
                 Mux(wb_ctrl.mul, mul.map(_.io.resp.bits.data).getOrElse(wb_reg_wdata),
                 wb_reg_wdata))))
```    
I searched over the codebase in the RocketCore, and found nothing to handle this , it seems that the second operation's write wont take effect.  
Can anyone shed some light into this?  when will the second write take effect? 
@aswaterman Thanks

aswaterman, 2021-10-16:
Please supply a failing test case if you think there's a bug.

jerryhethatday, 2021-10-17:
> Please supply a failing test case if you think there's a bug.

Thanks for your replay. I didn't mean there is a bug. I actually wanted to know how it works. According to wdata logic, seems that 
io.dmem.resp.bits.data(xLen-1, 0) will be written to the reg file instead of  wb_reg_wdata under circumstances I described. I searched the code base(rocketcore.Scalia)and found nothing to handle this. 
Simply put, I know this is right because you guys are super smart, but I am a rookie and have trouble to figure that out so far. 
Thanks for your work again!

aswaterman, 2021-10-17:
If a load writeback occurs at an unexpected time, the `replay_next` signal is asserted on the previous cycle, so that there will be no conflict in arbitrating for the write port.

jerryhethatday, 2021-10-18:
> If a load writeback occurs at an unexpected time, the `replay_next` signal is asserted on the previous cycle, so that there will be no conflict in arbitrating for the write port.

Got it， I did not read the DCache Module yet， Sorry for my inattention, thanks agagin!

closed 2021-10-18 by jerryhethatday