chipsalliance/rocket-chip #2902: Why does CLINT's timecmp have no reset?

michael-etzkorn, 2021-11-04:
Currently, reset leaves timecmp untouched which can cause issues with some RTOS that rely on time comparison for OS ticks. 
https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/devices/tilelink/CLINT.scala#L96 

Is the rationale behind not resetting timecmp that even though time resets, the device doesn't know what we want to reset timecmp to? Shouldn't it reset it to 0 and allow the trap handler to update it as opposed to requiring initialization in the firmware?

michael-etzkorn, 2022-03-25:
Moving question to stack overflow.

closed 2022-03-25 by michael-etzkorn

a0u, 2022-03-25:
Regardless of the initial value of `timecmp`, there must be some privileged code separate from the IRQ handler to set `MTIE`/`STIE`. At that point, it's not an onerous requirement for software to increment `mtimecmp`/`stimecmp` to deterministically schedule the first timer event.

It's unclear to me how relying on `timecmp` being reset to 0 and immediately taking a timer interrupt after enabling them would be appreciably simpler or more beneficial.

michael-etzkorn, 2022-03-25:
Certainly not onerous and I believe I understand the rationale. An example OS case on reset for leaving `mtimecmp` untouched would help cement the reasoning for me.

> It's unclear to me how relying on timecmp being reset to 0 and immediately taking a timer interrupt after enabling them would be appreciably simpler or more beneficial.

I think the simplicity comes in one making the implicit assumption `mtimecmp` resets when porting an RTOS and only find out through debugging that this isn't the case. Especially considering it resets to 0 during power-on anyway.