chipsalliance/rocket-chip #2910: Could use a minimal example of how to generate output from a Config

michael-etzkorn, 2021-11-25:
As stands, this explains CDE and the config system well, but it's still unclear from a novice perspective how we can generate an output Verilog file from our stand-alone config. The flows in rocket-chip and chipyard are a little bit more complex but are probably the best-known examples currently. It'd be nice to have smaller examples and information in the readme about generating output from a config.

sequencer, 2021-11-25:
CDE is only a parameter module, which can be used in any Scala software, so I think this isn't a cde question, but rocket-chip question, so I transfer the issue here for further recording.

michael-etzkorn, 2021-11-25:
I guess I should reframe this issue to the more general case. I'm not clear on how a config can be executed to populate the parameter values, and perhaps this is what needs a minimal example in the api-chipsalliance-config readme (or a pointer to one).

michael-etzkorn, 2021-11-25:
Upon rechecking the `api-chipsalliance=config` README. It does seem this minimal example exists. I now see how parameters is populated from a declared config. Thanks for making me think about it in the more general sense Jiuyang!

closed 2021-11-25 by michael-etzkorn