chipsalliance/rocket-chip #2917: when  'git submodule update --init --recursive', Cloning into '/xxx/xxx/rocket-tools/fsf-binutils-gdb'... is   failed

lpppeipei, 2021-12-14:
**Type of issue**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

michael-etzkorn, 2021-12-27:
I imagine this can be fixed by following the `README`'s instruction for tool-setup

https://github.com/chipsalliance/rocket-chip#setting-up-the-riscv-environment-variable

michael-etzkorn, 2022-10-15:
Closed due to lack of activity and being user specific.

closed 2022-10-15 by michael-etzkorn