chipsalliance/rocket-chip #2938: when will the scoreboard(core) be cleared in term of a dcache(blocking) refill?

jerryhethatday, 2022-02-23:
When a dcache miss happens,  the wb_dcache_miss (`val wb_dcache_miss = wb_ctrl.mem && !io.dmem.resp.valid`) will be asserted, and therefore  specific bit of the scoreboard will be set`(val wb_set_sboard = wb_ctrl.div || wb_dcache_miss || wb_ctrl.rocc).`  However, I have trouble figuring out the time that bit gets cleared once the missing block is acquired. As far as I can tell, the bit in scoreboard will be cleared only when `llwen` is asserted(`sboard.clear(ll_wen, ll_waddr)`),  however the `llwen` is set to `true` when `dmem_resp_replay && dmem_resp_xpu` is asserted , seems only the mmio resp will set the `io.dmem.resp.bits.replay` and consequently make `llwen` asserted.:
```  when (doUncachedResp) {
    assert(!s2_valid_hit)
    io.cpu.resp.bits.replay := true
    io.cpu.resp.bits.addr := s2_uncached_resp_addr
  }
```
Can anyone shed some light into this? Is there something that I miss? Thanks
@aswaterman

aswaterman, 2022-02-23:
The scoreboard is only set for instructions that actually retire.  In the event of a blocking miss, the offending memory access doesn't actually retire (it's nacked and replayed).  So the scoreboard never gets set (and therefore never gets cleared).

jerryhethatday, 2022-02-24:
> The scoreboard is only set for instructions that actually retire. In the event of a blocking miss, the offending memory access doesn't actually retire (it's nacked and replayed). So the scoreboard never gets set (and therefore never gets cleared).

@aswaterman  Thanks for your reply.  I know the general idea, which the offending mem access instruction will get replayed. But specifically I thought the `wb_dcache_miss ` will be asserted when the dcache miss happens at `s2` of dcache : 
1: Dcache deasserts the io.cpu.resp.valid  at s2 when there is a cache miss `io.cpu.resp.valid := (s2_valid_hit_pre_data_ecc || doUncachedResp) && !s2_data_error`
2: Consequently `wb_dcache_miss ` in the core will be asserted : `val wb_dcache_miss = wb_ctrl.mem && !io.dmem.resp.valid`, which will assert  the wb_set_sboard:  `val wb_set_sboard = wb_ctrl.div || wb_dcache_miss || wb_ctrl.rocc`, then the scoreboard will get set : `sboard.set(wb_set_sboard && wb_wen, wb_waddr)`
I know there is something I miss, but I just have no clue what is that. Thank you so much!

aswaterman, 2022-02-24:
As you said, `io.cpu.resp.valid` is deasserted.  But that means `wb_dcache_miss` will _not_ be asserted, because it depends upon `io.dmem.resp.valid`, which is connected to the deasserted `io.cpu.resp.valid` signal.

I'm guessing you aren't looking at a waveform, since you probably would've figured this out already if you had been.  I'm also a proponent of studying the code first, then studying the waveform if that approach fails.  But it turns out that both approaches are useful.