chipsalliance/rocket-chip #2943: verilog.c:308:10: fatal error: verilog.h: No such file or directory

pcotret, 2022-03-07:
**Type of issue**: bug report 
**Impact**:  unknown
**Development Phase**: proposal
**Other information** : `rocket-tools` have been compiled with GCC 9 as GCC 10 didn't work... (got an issue on `riscv-openocd` https://github.com/riscv/riscv-openocd/issues/480)

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
```bash
git clone https://github.com/chipsalliance/rocket-chip
cd rocket-chip
cd emulator
make
``` 

**What is the current behavior?**
Got an error quite quickly during the model compilation:
```
In file included from ../V3ParseGrammar.cpp:27:
verilog.c:308:10: fatal error: verilog.h: No such file or directory
compilation terminated.
```
Same issue as https://github.com/chipsalliance/rocket-chip/issues/2766, but downgrading Bison to the 3.5.1 version didn't work in my case...

**What is the expected behavior?**
Verilator model being compiled successfully

**Please tell us about your environment:**

closed 2022-03-07 by pcotret

SeddonShen, 2023-09-29:
> **Type of issue**: bug report **Impact**: unknown **Development Phase**: proposal **Other information** : `rocket-tools` have been compiled with GCC 9 as GCC 10 didn't work... (got an issue on `riscv-openocd` [riscv/riscv-openocd#480](https://github.com/riscv/riscv-openocd/issues/480))
> 
> **If the current behavior is a bug, please provide the steps to reproduce the problem:**
> 
> ```shell
> git clone https://github.com/chipsalliance/rocket-chip
> cd rocket-chip
> cd emulator
> make
> ```
> 
> **What is the current behavior?** Got an error quite quickly during the model compilation:
> 
> ```
> In file included from ../V3ParseGrammar.cpp:27:
> verilog.c:308:10: fatal error: verilog.h: No such file or directory
> compilation terminated.
> ```
> 
> Same issue as #2766, but downgrading Bison to the 3.5.1 version didn't work in my case...
> 
> **What is the expected behavior?** Verilator model being compiled successfully
> 
> **Please tell us about your environment:** - version: `git commit e2c6d1577a75f506fe992c3eb20a75174504476e` - OS: `Linux debian 5.10.0-11-amd64 #1 SMP Debian 5.10.92-1 (2022-01-18) x86_64 GNU/Linux`

Excuse me, I wanna to know how did you slove this problem, i was in the same case...

khoipkv, 2025-06-24:
Any luck? I'm also getting the same issue

CColombus, 2025-11-07:
For anyone stuck on the same issue, the solution from [#2766](https://github.com/chipsalliance/rocket-chip/issues/2766) works for bison 3.5.1

Just remember to remove you current bison version and add the new installation_directory/bison/bin to PATH before retrying. Also do a `make distclean` on verilator directory if needed.