chipsalliance/rocket-chip #2951: fatal: repository 'https://anongit.freedesktop.org/pixman/' not found

jerryhethatday, 2022-03-23:
I am tring to `git submodule update --init --recursive` to update the submodule following instructions below:
```
$ git clone https://github.com/freechipsproject/rocket-tools
$ cd rocket-tools
$ git submodule update --init --recursive
$ export RISCV=/path/to/install/riscv/toolchain
$ export MAKEFLAGS="$MAKEFLAGS -jN" # Assuming you have N cores on your host system
$ ./build.sh
$ ./build-rv32ima.sh (if you are using RV32).
```
But it display the following error messages:
```

Cloning into '/home/riscv/Documents/code/riscvtoolchian/rocket-tools/riscv-gnu-toolchain/riscv-qemu/pixman'...
fatal: repository 'https://anongit.freedesktop.org/pixman/' not found
fatal: clone of 'git://anongit.freedesktop.org/pixman' into submodule path '/home/riscv/Documents/code/riscvtoolchian/rocket-tools/riscv-gnu-toolchain/riscv-qemu/pixman' failed
Failed to clone 'pixman'. Retry scheduled
Cloning into '/home/riscv/Documents/code/riscvtoolchian/rocket-tools/riscv-gnu-toolchain/riscv-qemu/pixman'...
fatal: repository 'https://anongit.freedesktop.org/pixman/' not found
fatal: clone of 'git://anongit.freedesktop.org/pixman' into submodule path '/home/riscv/Documents/code/riscvtoolchian/rocket-tools/riscv-gnu-toolchain/riscv-qemu/pixman' failed
Failed to clone 'pixman' a second time, aborting
Failed to recurse into submodule path 'riscv-gnu-toolchain/riscv-qemu'
Failed to recurse into submodule path 'riscv-gnu-toolchain'

```
I googled this issue using key word"fatal: repository 'https://anongit.freedesktop.org/pixman/' not found", and Found this :https://github.com/sifive/freedom/issues/66  So, I typed  the following in the terminal:
`git config --global url."https://git.qemu.org/git".insteadOf https://git.qemu-project.org` However, the error remains. 
Can anyone shed some light into this? 
Thanks!
@mwachs5 @sequencer

michael-etzkorn, 2022-03-26:
Looking at `riscvtoolchian` with ai swapped, could it be a typo to your riscv-tool-chain directory?

jerryhethatday, 2022-03-27:
I tried many times cloning the tool chain following what the rocket chip repo instructed. But it doesn't work, I am kind of sure that the tool chain submodule in rocketchip is outdated(pixman can not be found).
I then take a detour using chipyard repo to set up the tool chain environment and RISCV environment variable is correctly set up by running the env.sh. Finally, the rockitchip repo works!
Anyway, I think rocket repo should update the readme instructions. Some part of it is outdated. Will close this.

jerryhethatday, 2022-03-27:
> Looking at `riscvtoolchian` with ai swapped, could it be a typo to your riscv-tool-chain directory? 

Thanks. That is a typo. But that's not causing the issue. Thanks for your carefulness.

closed 2022-03-27 by jerryhethatday