chipsalliance/rocket-chip #2953: Refactor Rocket Chip test framework

sequencer, 2022-03-25:
Most of configurations in RC is untested. Making RC almost impossible to accept RTL changes to new RV extension from community. 
So here I wanna request a new test framework:
1. A Scala `TestFramework` based test framework.
Currently RC uses Makefile, making Scala guys hard to add new test. It's reasonable to migrate most of test to Scala `TestFramework`, `ScalaTest` or `uTest` are both OK.  
2. Get rid of dependency to SW, upstream compiler is almost mature for RISC-V now, user don't need to compile their toolchain at 2022, directly depends to LLVM seems reasonable.
3. Use test case from
https://github.com/riscv-software-src/riscv-tests
https://github.com/riscv-non-isa/riscv-arch-test
4. frameworks for different developing levels
- a spike-cosim library is required for debug, it should support verilator at least.
- JTAG bitbang should be replaced to JTAG VPI to speed up binary loading.

**Type of issue**: feature request

**Impact**: no functional change

**Development Phase**: request

michael-etzkorn, 2022-03-25:
@sequencer this is maybe something we can discuss April 6th, but I believe for (4), it'd be reasonable to consider (and I'd be willing to help develop) UVM-based or generic SystemVerilog testbenches for testing the Verilog IR.

sequencer, 2022-04-06:
>  it'd be reasonable to consider (and I'd be willing to help develop) UVM-based or generic SystemVerilog testbenches for testing the Verilog IR.

Thanks @michael-etzkorn!
I think before ventilator(or any other Open-Source high-performance simulator) providing the full UVM support. We should continue stick in the DPI/VPI for the best compatibility.
And also there is a needs to test RC with commercial tools(like VCS+Verdi). Design with GTKWave is painful.
Currently the RC do support a DPI-based simulation, see LazyModules prefix with `Sim`(`SimJTAG`, `SimDTM`) etc...
And it works good with Verilator and VCS, but needs more documentation and standardization.

So basically, from my point of view: the "test framework" should be a refactor to current test framework. I think we needs hands on it, we can discuss on the RCW meeting!