chipsalliance/rocket-chip #2955: Unable to get Rocket Chip waveforms for GTKwaves

Ryabkomv, 2022-04-02:
I want to run a program on Rocket core and observe all the signals in corresponding registers in GTKwave (e.g. PC, register file, ALU registers and wires etc.)

However, the only I get (both in chipyard and rocket chip) is some strange list of wires in GTKwave, which I cannot relate to the core/tile.

I followed instructions at [https://github.com/chipsalliance/rocket-chip](https://github.com/chipsalliance/rocket-chip) for installations. I'm able to run:

```
make CONFIG = freechips.rocketchip.system.TinyConfig (I did DefaultConfig as well)
make verilog 
etc.
```
Also I generate *.vcd files with

`make run-debug CONFIG= freechips.rocketchip.system.TinyConfig`

or specific file with hello word.

For each file there is corresponding *.out file with all executed instructions, so I naively think that I can run any of such generated *.vcd and see all the register states for all instructions.

However I get only [strange wires like here](https://i.stack.imgur.com/H0sif.png)

Elsewhere people demonstrate [reasonable signals like this:](https://i.stack.imgur.com/NhfTT.png)

I observe at the last image TestBench group. I did test benches for modelsim with pure verilog written in Quartus. However for rocket chip framework and Verilator in particular, I think I can run any *.vcd file.

It looks for me the same approach used in [theses](https://tigerprints.clemson.edu/cgi/viewcontent.cgi?article=4602&context=all_theses) page 26 with reasonable waves at page 27

Can somebody give me a hint, what is wrong with my approach?

michael-etzkorn, 2022-04-05:
> However I get only strange wires like here

That's the hierarchy view. You can use signal search to find the signals and append them to the waveform viewer. 
http://gtkwave.sourceforge.net/gtkwave.pdf#page=54

Ryabkomv, 2022-04-07:
Hello Jerry,
thanks for helping me. Actually in ldut there is nothing useful either.
[image]

jerryz123, 2022-04-07:
`ldut` is `RocketSubsystem`. I believe if you dig in `ldut/tile_prci_domain` a little further you'll find `RocketTile`

Ryabkomv, 2022-04-07:
Oh, how deep it is!
Thanks Jerry, I found something reasonable.
Jerry, may be you know, is there documentation with detailed description of signals in Rocket-Chip generator. Something like [this](https://docs.boom-core.org/en/latest/sections/instruction-fetch-stage.html). Basically I want to understand what are all these signals in hierarchy starting from TOP node without digging into Scala code.
Lot of thanks to you!

michael-etzkorn, 2022-04-20:
Unfortunately, the current recommended method for understanding the description of all the signals is by reading the Scala code or through exploration in simulation. #2960 is our open issue for improving documentation on things like signal names. 

I'll go ahead and close this since the question about hierarchy seems to be answered.

closed 2022-04-20 by michael-etzkorn