chipsalliance/rocket-chip #2960: [WIP] Documentation Plan

sequencer, 2022-04-08:
**Type of issue**: feature request

**Impact**: no functional change

**Development Phase**: request

**Other information**

michael-etzkorn, 2022-05-03:
A small list of things to provide quality documentation for: 
- FPGA and ASIC flow resources
- Configurations and Parameters
- RC's microarchitecture. Doc similar to BOOM's. 
    - helps users with situations like #2955 
- Running unit tests
- Running benchmarking 
- RoCC
- Bus Topology and Bus APIs. e.g.:
    - Diplomatic Widgets 
    - Location API
- Device tree
- Version releases - PRs should correlate to version updates and there should be patch notes with each release

rstarc, 2023-08-19:
What is the status of this issue? I would like to contribute
I see that there are multiple seemingly related branches named `*_doc` containing unmerged commits. Is anyone actively else actively working on this?