chipsalliance/rocket-chip #2971: [Bug Report] The width of tdata2 is less than XLEN

Phantom1003, 2022-05-02:
**Type of issue**: bug report
**Impact**: unknown
**Development Phase**: proposal
**Other information**
In the following test case, we write 0x2000000100004001 to tdata2 with a valid value in high bits, compared to the result from spike, we found that rocket ignores the 0x2 in the highest byte.
```
[rocket] 3 0x00800001c4 (0x00c5e633) x12 0x2000000100004001
[spike]  core   0: 0x00000000800001c4 (0x00c5e633) or      a2, a1, a2
[rocket] 3 0x00800001c8 (0x7a261073)
[spike]  core   0: 0x00000000800001c8 (0x7a261073) csrw    tdata2, a2
[rocket] 3 0x00800001cc (0x7a2025f3) x11 0x0000000100004001
[spike]  core   0: 0x00000000800001cc (0x7a2025f3) csrr    a1, tdata2
[error] WDATA EMU 2000000100004001 , DUT 0000000100004001
```

[rocket-4.zip](https://github.com/chipsalliance/rocket-chip/files/8603173/rocket-4.zip)

According to the debug specification, the width of tdata2 should be XLEN bits, but we found that the width in rocket is less than XLEN, which may trigger unintended breakpoints or produce incorrect comparison results.

**Please tell us about your environment:**
    - version: edc8d40

aswaterman, 2022-05-04:
Similar to #2975 and #2976, this shouldn't be a bug, but I'm not an expert on the debug spec, so I can't point you to the chapter and verse that permits this behavior.  Presumably there is a statement that `tdata2` is WARL and need not hold all 2^XLEN possible values.  cc @ernie-sifive

closed 2022-05-04 by Phantom1003

sequencer, 2022-05-04:
@ernie-sifive I saw current Debug Spec seems removed:
> This trigger type may be limited to address comparisons (select is always 0) only. If that is the case, then tdata2 must be able to hold all valid virtual addresses but it need not be capable of holding other values.

https://github.com/riscv/riscv-debug-spec/blob/17187e4002a25a336a74ba3ac90aa2796d8ff500/xml/hwbp_registers.xml#L102

I wonder the design purpose about this.
Does this also means if upgrading to Debug Spec 1.0, we should make `tdata2` `XLEN` size?

ernie-sifive, 2022-05-04:
@sequencer The text is still present (though modified) in the current Stable Debug Spec: 

https://github.com/riscv/riscv-debug-spec/blob/17187e4002a25a336a74ba3ac90aa2796d8ff500/xml/hwbp_registers.xml#L246

Basically, `tdata2` needs to be able to hold values needed by the types of triggers implemented.  rocket-chip only supports type=2 and select=0 and does not support masking so although `tdata2` is XLEN bits wide, not all of the bits necessarily need to be writable.