chipsalliance/rocket-chip #2976: [Bug Report] xepc does not compress the written values

Phantom1003, 2022-05-04:
**Type of issue**: bug report
**Impact**: unknown
**Development Phase**: proposal
**Other information**
In the following test case, we write 0x2000000080000000 to mepc, which is an illegal address out of the virtual address range.
Since the pc register in rocket is less than 64-bit, rocket has a compress function to compress the 64-bit address to 40-bit.
https://github.com/chipsalliance/rocket-chip/blob/be37886d27592b010d30bb87cfb81283575b4e96/src/main/scala/rocket/RocketCore.scala#L1032-L1038
We found that rocket ignores this step when writing to xepc, rocket direct return to 0x80000000.
```
[rocket] 3 0x00800001a4 (0x00a5e5b3) x11 0x2000000080000000
[spike]  core   0: 0x00000000800001a4 (0x00a5e5b3) or      a1, a1, a0
[rocket] 3 0x00800001a8 (0x34159073)
[spike]  core   0: 0x00000000800001a8 (0x34159073) csrw    mepc, a1
[rocket] 3 0x00800001ac (0x30200073)
[spike]  core   0: 0x00000000800001ac (0x30200073) mret
[spike]  0 0x0080000000 (0x0540006f)
[spike]  core   0: exception trap_instruction_access_fault, epc 0x2000000080000000
[spike]  core   0:           tval 0x2000000080000000
[spike]  core   0: 0x0000000080000004 (0x34302f73) csrr    t5, mtval
[error] PC SIM 0000000080000004 , DUT 0000000080000000
[error] INSN SIM 34302f73 , DUT 0540006f
```
[rocket-6.zip](https://github.com/chipsalliance/rocket-chip/files/8618568/rocket-6.zip)

**Please tell us about your environment:**
    - version: edc8d40

aswaterman, 2022-05-04:
This is not a bug; the ISA explicitly permits it: https://github.com/riscv/riscv-isa-manual/blob/55cd85c891f21302cdd9d1bfa60ee87a7a241f93/src/supervisor.tex#L636-L637

Phantom1003, 2022-05-04:
```
Prior to writing mepc, implementations may convert an invalid address into some other invalid address that mepc is capable of holding.
```
I mean here rocket directly truncates written value, instead of converting it to an invalid address.

aswaterman, 2022-05-04:
That sentence refers to implicit writes of mepc (i.e., on a trap).  It's perhaps unclearly worded; it means to say that "When an implementation writes mepc, it may convert [...]", as opposed to "when software writes mepc".

Phantom1003, 2022-05-04:
Under normal use scenarios, it really has no problem.
From the security view, attackers may use it to inject program control flow, suppose the attack want to jump to 0x80000000.
With the simply truncate writes, he only needs to construct 0x????????80000000 is sufficient.
Since the compress function will generate an illegal address when feeding an address with random top bits, with this he need to construct the complete address 0x0000000080000000.

aswaterman, 2022-05-04:
If the attacker has control over what gets written into mepc, the system is quite badly compromised.  The incremental security benefit here seems questionable.  In any case, it seems like an argument about what the ISA allows, rather than an argument about what Rocket does.

Phantom1003, 2022-05-04:
I think ISA should not need to limit what can be written, I just wanted to point out that rocket protects the normally generated pc (e.g. jalr) with compression and ignores the possible pc generated in csr.
If you think this is not a problem, I will leave it.

aswaterman, 2022-05-04:
FWIW, the invalid->invalid recoding requirement on implicit writes wasn't motivated by security.  For *tval, we mandated it so that software can determine whether or not to service a page fault (if the address is invalid, software shouldn't attempt to service a fault).  For both *tval and *epc, it aids debugging.

Phantom1003, 2022-05-04:
But different from *tval, *epc accepts the user input and will be used during *ret.
And the compression function does ensure security by converting illegal user state addresses into kernel state addresses, that is why I think it is necessary to compress the written values for the csrw instruction as well.

closed 2022-05-04 by Phantom1003