chipsalliance/rocket-chip #2980: [Bug Report] incorrect xs field check under VS

Phantom1003, 2022-05-08:
**Type of issue**: bug report
**Impact**: unknown
**Development Phase**: proposal
**Other information**
In the following test case, we disable xs field in vsstatus. Spike throws an illegal instruction exception, while rocket still executes as normal.
```
[rocket] 1 0x0008000019c (0x20002573) x10 0x0000000200000000
[spike]  core   0: 0x000000008000019c (0x20002573) csrr    a0, vsstatus
[rocket] 1 0x000800001a0 (0x54c5f50b) x10 p10 0xXXXXXXXXXXXXXXXX
[spike]  core   0: 0x00000000800001a0 (0x54c5f50b) custom0.rd.rs1.rs2 (args unknown)
[spike]  core   0: exception trap_illegal_instruction, epc 0x00000000800001a0
[spike]  core   0:           tval 0x0000000054c5f50b
[spike]  core   0: 0x0000000080000004 (0x34302f73) csrr    t5, mtval
[error] PC SIM 0000000080000004, DUT 00000000800001a0
[error] INSN SIM 34302f73, DUT 54c5f50b
[CJ] Commit Failed
```

[rocket-7.zip](https://github.com/chipsalliance/rocket-chip/files/8646300/rocket-7.zip)

We believe there is a typo in the rocc_illegal signal, where the `.vs` should be `.xs`:
https://github.com/chipsalliance/rocket-chip/blob/850e1d5d56989f031fe3e7939a15afa1ec165d64/src/main/scala/rocket/CSR.scala#L854

**Please tell us about your environment:**
    - version: edc8d40

Phantom1003, 2022-05-10:
@aswaterman Would you please confirm this report?

aswaterman, 2022-05-11:
I would have said this differently: RoCC and Hypervisor extension are currently incompatible.  This is a request for a new feature, not a report about a bug.

Please work with the folks at UC Berkeley who use RoCC to make sure they are OK with whatever you propose.  I don't work with RoCC.  (cc @jerryz123 to see who should work with you on this PR.)

Phantom1003, 2022-05-11:
I still insist that this is a bug, and this report contains two problems.

1. The xs bits of vsstatus is writable, which is inconsistent with the specification:
> Read-only fields SD and XS summarize the extension context status as it is visible to VS-mode
only. For example, the value of the HS-level sstatus.FS does not affect vsstatus.SD.

https://github.com/chipsalliance/rocket-chip/blob/850e1d5d56989f031fe3e7939a15afa1ec165d64/src/main/scala/rocket/CSR.scala#L1337-L1347

2. The rocc_illegal signal incorrectly uses vs field, which is used for vector_illegal. And if RoCC and Hypervisor extension are currently incompatible, there should be `reg_mstatus.v && usingRoCC` instead of `reg_vsstatus.vs`.
https://github.com/chipsalliance/rocket-chip/blob/850e1d5d56989f031fe3e7939a15afa1ec165d64/src/main/scala/rocket/CSR.scala#L851-L854

aswaterman, 2022-05-11:
I agree XS should not be directly writable, but it’s only a bug that manifests with RoCC enabled, and so the maintainers/users of RoCC (i.e. not me :-)) should deal with it.

I also agree that rocc_illegal should use XS, not VS, but this is really an unrelated topic. Please keep bug reports to one bug at a time.

jerryz123, 2022-05-11:
The last time this came up was 2 years ago. This issue gives me deja-vu.

I believe XS should be read-only, but hard wired to dirty when RoCC is present. https://github.com/chipsalliance/rocket-chip/pull/2508

aswaterman, 2022-05-11:
I agree that’s the lowest-effort-but-seemingly-correct thing to do. It also makes it obvious what to do with vsstatus.

Phantom1003, 2022-05-14:
Thanks to Andrew and Jerry, the bugs mentioned have been fixed.

closed 2022-05-14 by Phantom1003