chipsalliance/rocket-chip #2981: How to rightly rotate the graphml layout

jerryhethatday, 2022-05-10:
It seems not a good place to post this problem here . But I know the rocketchip community deals with this problem either. 
I opened the generated diplomacy connection graph in yEd version 3.22, and It's a mess
[image]

I then use the one-click layout feature to auto layout this, it worked just fine:
[image]
However, the layout direction is just wrong, I need to rotate it 90degree clokcwisely, I searched online, and found the following option:
[image]
However, after this operation, seems that lots of the elements are not well aligned and it's a mess.
[image]
So, my problem is simple, hot can I check this graph rightly, sorry for the overkill of posting an issue here.

jerryhethatday, 2022-05-10:
@hcook

sequencer, 2022-05-10:
The generated `graphml` doesn't contain any layout informations. that should be an issue to your `graphml` viewer(I bet you are using `yEd`), for simplest way, you can directly export it to `pdf` and rotate which via your pdf viewer.

closed 2022-05-10 by sequencer