chipsalliance/rocket-chip #2987: [Bug Report] allow to select both hypervisor extension and RV32E

Phantom1003, 2022-05-14:
**Type of issue**: bug report
**Impact**: unknown
**Development Phase**: proposal
**Other information**
The following dut was generated with RV32E, H ext, none FPU, the value in misa also proves this.

```
[rocket] 3 0x00080000000 (0x0480006f)
[spike]   core   0: 0x0000000080000000 (0x0480006f) j       pc + 0x48
[rocket] 3 0x00080000048 (0x301022f3) x 5 0x8000000000941095
[spike]   core   0: 0x0000000080000048 (0x301022f3) csrr    t0, misa
[rocket] 3 0x0008000004c (0x20001073)
[spike]   core   0: 0x000000008000004c (0x20001073) csrw    vsstatus, zero
[rocket] 3 0x00080000050 (0x200022f3) x 5 0x0000000200000000
[spike]   core   0: 0x0000000080000050 (0x200022f3) csrr    t0, vsstatus
...
[rocket] 3 0x0008000008c (0x00000793) x15 0x0000000000000000
[spike]   core   0: 0x000000008000008c (0x00000793) li      a5, 0
```

[rocket-8.zip](https://github.com/chipsalliance/rocket-chip/files/8691872/rocket-8.zip)

The specification says that the H extension depends on the I set instead of E set, which rocket does not check at generation time:
> The hypervisor extension depends on an “I” base integer ISA with 32 x registers (RV32I or RV64I), not RV32E, which has only 16 x registers.

**Please tell us about your environment:**
    - version: edc8d40

Phantom1003, 2022-05-14:
cc to @aswaterman

jerryz123, 2022-05-14:
Wouldn't it be sufficient to add the check to validate `!(usingHypervisor && useRVE)`?

aswaterman, 2022-05-14:
Hypervisor + RVE is an illogical configuration; the appropriate bug fix is to install a require() to the contrary.

Phantom1003, 2022-05-14:
The patch has been merged, thanks for the quick replies!

closed 2022-05-14 by Phantom1003