chipsalliance/rocket-chip #2989: how to load a data file into a SRAM

JACKLIAO0, 2022-05-14:
**Type of issue**: questions

**Detail information**
 
I want to load a data file into a SRAM of RocketChip.

I have tried the following：

1. I have generated a SRAM on cbus like this:
trait HasTestRAM { this: BaseSubsystem => val testRAM = LazyModule( new TLRAM(AddressSet(0x40000000, 0x7FFFF), beatBytes = cbus.beatBytes) ) testRAM.node := cbus.coupleTo("sram") { TLFragmenter(cbus.beatBytes,4096) := _ } }

2. put HasTestRAM in ExampleRocketSystem:
class ExampleRocketSystem(implicit p: Parameters) extends RocketSubsystem
    with CanHaveAXI4RailBus
    with HasAsyncExtInterrupts
    with CanHaveMasterAXI4MemPort
    with CanHaveMasterAXI4MMIOPort
    with CanHaveSlaveAXI4Port
    **with HasTestRAM**

3. I have used loadMemoryFromFileInline API as to generate readmemh in generated verilog code, add the following code in ExampleRocketSystemModuleImp
 `**loadMemoryFromFileInline(ldut.RailBusAXI4RAM.module.mem, "xxx.txt")**`

But after I run **make verilog** in vsim direction, I can't find ”readmemh“ in generated verilog . Is there something wrong with my way of using this loadMemoryFromFileInline ?

Maybe there are other ways to load memory from file?

@michael-etzkorn @sequencer

michael-etzkorn, 2022-05-14:
Instead of scoping mem in the `ExampleRocketSystemModuleImp`, can you try calling `loadMemoryFromFileInline` from where `val mem` is instantiated? I'm guessing you have to be in the context of the module for the API. 

- - - 
- As an aside, RC should consider using `loadMemoryFromFileInline` for BootROM since BootROM builds really, _really_ slow on large binaries.  https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/devices/tilelink/BootROM.scala#L44

JACKLIAO0, 2022-05-15:
> Instead of scoping mem in the `ExampleRocketSystemModuleImp`, can you try calling `loadMemoryFromFileInline` from where `val mem` is instantiated? I suspect you have to be in the context of the module for the API to work.
> 
> * As an aside, RC should consider using `loadMemoryFromFileInline` for BootROM since BootROM builds really, _really_ slow on large binaries.  https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/devices/tilelink/BootROM.scala#L44

@michael-etzkorn 
Thanks for you reply.

After I researched the usage of loadMemoryFromFileInline, I found the reasons for the above is:
[https://github.com/chipsalliance/chisel3/issues/2532#issue-1236237905](url)

But I still don't have a good way to accomplish my request. I want to initiate SRAM from a data file.

sequencer, 2022-05-15:
Generally this is a bug of Chisel and FIRRTL, you need to submit a bug report there.

This is because:
1. RocketChip depends on the [`firrtl.passes.ReplaceMemMacros`](https://github.com/chipsalliance/firrtl/blob/37c8528cfed4395924820b54498ef761ded17393/src/main/scala/firrtl/passes/memlib/ReplaceMemMacros.scala
) to replace the behavior memory to `ExtModule` and generate a config to memory compiler.
2. The [dependency](https://github.com/chipsalliance/chisel3/blob/2b48fd15a7711dcd44334fbbc538667a102a581a/src/main/scala/chisel3/util/experimental/LoadMemoryTransform.scala#L203-L204
) of `chisel3.util.experimental.LoadMemoryTransform` may not correct. which may need add the `optionalPrerequisiteOf` of `firrtl.passes.ReplaceMemMacros`. Basically, `firrtl.passes.ReplaceMemMacros` should ignore the memory of `chisel3.util.experimental.LoadMemoryTransform`, or schedule `chisel3.util.experimental.LoadMemoryTransform` before `firrtl.passes.ReplaceMemMacros`.
Since `firrtl.passes.ReplaceMemMacros` will replace all memory of the circuit, which contains the memory which needs load data from file. after running the `firrtl.passes.ReplaceMemMacros`, `chisel3.util.experimental.LoadMemoryTransform` cannot find the target memory.

closed 2022-05-15 by sequencer

michael-etzkorn, 2022-05-15:
Ah, that’s unfortunate that the ReplaceMemMacro eats the readmem. There’s
definitely still ways to do this with FIRRTL 1.5 and chisel 3.5, it’ll just
be a bit of a hack. Either reading the bin into a vec similar to what
BootROM.scala does, inline’d verilog,  disabling the FIRRTL men swap so the
default behavioral memory is emitted, or (probably the biggest hack)
hooking a script to add the readmem after the verilog is emitted.

JACKLIAO0, 2022-05-17:
Thanks for reply.

I will have a try using ways you have mentioned above.