chipsalliance/rocket-chip #2993: Code Anatomy of FrontEnd, including RAS, BTB and BHT.

jerryhethatday, 2022-05-30:
The frontend is for fetching instructions from memory, conducting branch and jump prediction(speculative changing of fetch flow), then feeding the instructions to an intermediate fetch queue from which the backend of cpu core can begin decoding the instruction stream. Frontend is somehow similiar to the traditional fetch stage with more powerful functionality added. It's ofthen the standard implementation of modern processor design. 
Specifically, the rocketchip implementation of frontend has 3 pipleline stages(`s0 s1 s2`):

### 1. S0
In `s0`, initiating the fetch request to sub module `icache` if there is a valid and explicit backend request(`io.cpu.req.valid`) or there are empty slots in the fetching queue(`s0_fq_has_space`: a ShiftQueue), the icache will normally return the valid resp 2 cycles later at **s2** . Also note that icache of RC is organized as VIPT,  therefore it needs physical addr to do tag matching, the vaddr<->paddr translation is done at `s1`
```
  val s0_fq_has_space =
    !fq.io.mask(fq.io.mask.getWidth-3) ||
    (!fq.io.mask(fq.io.mask.getWidth-2) && (!s1_valid || !s2_valid)) ||
    (!fq.io.mask(fq.io.mask.getWidth-1) && (!s1_valid && !s2_valid))
  val s0_valid = io.cpu.req.valid || s0_fq_has_space
  icache.io.req.valid := s0_valid
  icache.io.req.bits.addr := io.cpu.npc
  icache.io.invalidate := io.cpu.flush_icache
```
### 2. S1 
In `s1`,  the request to tlb is initiated to get the actual physical address(paddr), response will come at the same cycle:`s1`. This paddr then goes into `icache ` as the physical tag. It's fed into icache at `s1`:
```
val tlb = Module(new TLB(true, log2Ceil(fetchBytes), TLBConfig(nTLBSets, nTLBWays, outer.icacheParams.nTLBBasePageSectors, outer.icacheParams.nTLBSuperpages)))
tlb.io.req.valid := s1_valid && !s2_replay//hjr tlb req and normal resp all occured at s1.
tlb.io.req.bits.vaddr := s1_pc// hjr begin looking up tlb, the physical page number will be returned as the same cycle
tlb.io.req.bits.passthrough := Bool(false)//hjr not just for permission checking, also conduct address translation
tlb.io.req.bits.size := log2Ceil(coreInstBytes*fetchWidth)
tlb.io.sfence := io.cpu.sfence
//hjr the kill for tlb just tell tlb do not conduct tlb refill if there
//is a tlb miss, This just means if the request didn't make it to s2(!s2_valid), even though there is a tlb miss, ignore that: just don't refill it
tlb.io.kill := !s2_valid
icache.io.s1_paddr := tlb.io.resp.paddr
```
Note that  tlb may also report **miss** the same cycle it receives the request. Once miss happens, the ongoing icache request(now at `s1`) should be killed because there is no valid physical tag info to feed into `icache` at `s1`, once `icache` request is killed 1 cycle later (at `s1`), it won't report valid response(`val s2_valid = RegNext(s1_valid && !io.s1_kill, Bool(false))` and `io.resp.valid := s2_valid && s2_hit` ) at `s2`, therefore pc at `s2` won't pollute the `fq`:
```
icache.io.s1_kill := s2_redirect || tlb.io.resp.miss || s2_replay
//hjr note that icache.io.resp.valid won't be asserted if the icache req is killed at s1
fq.io.enq.valid := RegNext(s1_valid) && s2_valid && (icache.io.resp.valid || !s2_tlb_resp.miss && icache.io.s2_kill)
```
Besides tlb, request to tlb also happens at s1. In short summary(detailed clarifictions later), Branch Target Buffer is a cache storeing the mapping between address of conditional branches and the predicted target address. In RC, however, the BTB module is a little bit of textbook deviation: It stores target addr of all operations that change the fetch flow: unconditional branch(jal call, jalr call&return, normal jal&jalr) and  conditional branch.  Once there is a hit of `s1_pc` in BTB, the predicted target address will be sent to `s0` as next fetching address combinationally, the fetching flow will therefore be changed:
```
val btb = Module(new BTB)
btb.io.req.bits.addr := s1_pc//hjr btb requests are sent at s1
when (!s2_replay) {//hjr if the previous instruction should be replayed, .
  //hjr when s2 has a redirected instruction(or async from the backend), the one at s1 should not valid its BTB request because the one at s1 slot will be killed.
  btb.io.req.valid := !s2_redirect// hjr when no redirection happened at s2, btb request should be valid.
  s2_btb_resp_valid := btb.io.resp.valid
  s2_btb_resp_bits := btb.io.resp.bits
    }
when (btb.io.resp.valid && btb.io.resp.bits.taken) {
    predicted_npc := btb.io.resp.bits.target.sextTo(vaddrBitsExtended)//hjr predicted_npc<->s1
    predicted_taken := Bool(true)
}
```
There are some important signals to clarify here:

- s2_replay

This indicates whether the instruction at `s2` should perform a replay, scenarios when `replay `could happen at `s2` are

a): `icache` doesn't give a valid resp at `s2` (`icache.io.resp.valid`), meanwhile `s2_kill` of `icache`  is **not** asserted:(`icache.io.resp.valid || !s2_tlb_resp.miss && icache.io.s2_kill`)

b): if the shift queue is not ready for accepting new item: `!fq.io.enq.ready`

`s2_kill ` of icache just means _don't perform icache refill even if there is icache miss_ ; Reason of `s2_kill` assertion may be:

i):tlb reports _acess exception_ or _page fault_(`val s2_xcpt = s2_tlb_resp.ae.inst || s2_tlb_resp.pf.inst`)); 

ii):`s2_speculative && !s2_can_speculatively_refill`; Which means if you want to perform a speculative refill for icache with `disableSpeculativeICacheRefill `being asserted, the refill won't happen;   

Note that besides replay at frontend `s2`, there is `replay` signal fed into the fetch queue when there is icache parity error(`icache.io.resp.bits.replay`) or `icache.io.s2_kill && !icache.io.resp.valid && !s2_xcpt` is asserted. _I have trouble figuring out the rationale of feding this item into fetch queue and let the backend to handle replay logic._

Also note that the `s2_replay` should keep asserting if the previous `s2_replay` can not be absorbed by `s0` stage(` RegNext(s2_replay && !s0_valid, true.B)` while `val s0_valid = io.cpu.req.valid || s0_fq_has_space`) ; The corresponding code fragment is as follows:
```
[... 430 lines cut ...]
  idxPages(waddr) := idxPageUpdate +& 1 // the +1 corresponds to the <<1 on io.resp.valid
  tgtPages(waddr) := tgtPageUpdate
  cfiType(waddr) := r_btb_update.bits.cfiType
  isValid := Mux(r_btb_update.bits.isValid, isValid | mask, isValid & ~mask)
  if (fetchWidth > 1)
    brIdx(waddr) := r_btb_update.bits.br_pc >> log2Up(coreInstBytes)//hjr brIdx is caculated here instead of obtaining from FrontEnd module.

  require(nPages % 2 == 0)
  val idxWritesEven = !idxPageUpdate(0)

  def writeBank(i: Int, mod: Int, en: UInt, data: UInt) =
    for (i <- i until nPages by mod)
      when (en(i)) { pages(i) := data }

  //hjr updates the real page content at idxPageReplEn or tgtPageReplEn
  writeBank(0, 2, Mux(idxWritesEven, idxPageReplEn, tgtPageReplEn),
    Mux(idxWritesEven, page(r_btb_update.bits.pc), page(update_target)))
  writeBank(1, 2, Mux(idxWritesEven, tgtPageReplEn, idxPageReplEn),
    Mux(idxWritesEven, page(update_target), page(r_btb_update.bits.pc)))
  pageValid := pageValid | tgtPageReplEn | idxPageReplEn
}

```

There may be some extra clarifications, and I will post them in the comment below. But the depiction above just tells the general idea.

michael-etzkorn, 2022-06-04:
Hi Jerry,

Thanks for the write-up. Have you considered a blog style issue page like this one https://github.com/cisen/blog/issues in your own github repo? Alternatively, we can migrate these to a discussion board once we have that created (for now, we can back link this and #2992 by mentioning #2960 ). 

I'm spending some time cleaning up the issue page so it tracks legitimate tickets (e.g. bug issues and proposals). I'll leave the two tickets you've opened in open for now, but since they don't really fit the purpose of Github's Issue page, the plan will be to close them once I've gone through the back-log of old tickets.

jerryhethatday, 2022-06-07:
@michael-etzkorn  Hi Michael, Thanks for your reply.  I posted it here because I thought the RC repo is the very first place where people(jackaroo like me) ask for help. I know this kind of content doesn't fit the **issue** purpose. Sorry for this.
I am ok for posting in my own repo and mention [2960](https://github.com/chipsalliance/rocket-chip/issues/2960) there, but maybe fewer people will see the post.
A discussion board will be cool. 
For now, I will close this, people still can search within the repo and get the closed issue. This is totally ok for me. 
Thanks again for your work,

closed 2022-06-07 by jerryhethatday