chipsalliance/rocket-chip #2995: [Bug Report] fs is incorrectly set for illegal float instructions

Phantom1003, 2022-06-12:
**Type of issue**: bug report
**Impact**: unknown
**Development Phase**: proposal
**Other information**
In the following test case, we first set mstatus.fs to 01(initial), execute an illegal fld, and read mstatus again after returning from the trap.
From the co-simulation result, we find that rocket sets the fs field to 11(dirty).
```
[rocket] 3 0x00080000140 (0x00002537) x10 0x0000000000002000
[spike]  core   0: 0x0000000080000140 (0x00002537) lui     a0, 0x2
[rocket] 3 0x00080000144 (0x30052073)
[spike]  core   0: 0x0000000080000144 (0x30052073) csrs    mstatus, a0
[rocket] 3 0x00080000004 (0x34302f73) x30 0xfffffffffffffff8
[spike]  core   0: 0x0000000080000148 (0xff803487) fld     fs1, -8(zero)
[spike]  core   0: exception trap_load_access_fault, epc 0x0000000080000148
[spike]  core   0:           tval 0xfffffffffffffff8
[spike]  core   0: 0x0000000080000004 (0x34302f73) csrr    t5, mtval
...
[rocket] 3 0x00080000040 (0x30200073)
[spike]  core   0: 0x0000000080000040 (0x30200073) mret
[rocket] 3 0x0008000014c (0x300022f3) x 5 0x8000000a0001e080
[spike]  core   0: 0x000000008000014c (0x300022f3) csrr    t0, mstatus
[error] WDATA SIM 8000000a0001a080, DUT 8000000a0001e080
[error] check board clear 5 error
[CJ] integer register Judge Failed
```
The value from spike is 0x8000000a0001a080(fs = 01), while rocket is 0x8000000a0001e080(fs = 11).

We also compared cva6 and boom and found that they both did not set fs. 
We believe this problem has been missed because the current testing frameworks directly set fs to dirty, thus masking this change.

[rocket-9.zip](https://github.com/chipsalliance/rocket-chip/files/8886017/rocket-9.zip)

Phantom1003, 2022-06-12:
cc to @aswaterman and @jerryz123

aswaterman, 2022-06-13:
This isn’t actually a bug. The ISA permits that FS be set for any reason.

It probably is an error of intent, but fixing it might not be worth the effort, since it’s not invalid behavior.

aswaterman, 2022-06-13:
@jerryz123 should weigh in, though

Phantom1003, 2022-06-13:
Sorry, since the test program was randomly generated, I didn't notice that rocket uses a WARL rule for fs. 
In the test case given above, fs was actually set to dirty as early as when the value of mstatus.fs was set to initial.

All of the above is the case when rocket is configured with a false haveFSDirty.
I just tested havingFSDirty on and I found that rocket doesn't use the set_fs_dirty signal in the CSR module, so it will only set fs to dirty when accessing fcsr.
```
3 0x00080000158 (0x300022f3) x 5 0x8000000a0001a080
core   0: 0x0000000080000158 (0x300022f3) csrr    t0, mstatus
3 0x0008000015c (0x00002517) x10 0x000000008000215c
core   0: 0x000000008000015c (0x00002517) auipc   a0, 0x2
3 0x00080000160 (0xeac53087) f 1 p33 0xXXXXXXXXXXXXXXXX
core   0: 0x0000000080000160 (0xeac53087) fld     ft1, -340(a0)
f 1 p33 0x3fe0000000000000
3 0x00080000164 (0x300022f3) x 5 0x8000000a0001a080
core   0: 0x0000000080000164 (0x300022f3) csrr    t0, mstatus
[error] WDATA SIM 8000000a0001e080, DUT 8000000a0001a080
[error] check board clear 5 error
[CJ] integer register Judge Failed
```

After executing a fld, rocket still has an initial fs, which is obviously a bug.
I think we should disallow use to enable the haveFSDirty.

aswaterman, 2022-06-13:
That’s right, Rocket doesn’t support haveFSDirty, and it should probably have an assertion to that effect.

closed 2022-06-14 by Phantom1003