chipsalliance/rocket-chip #2996: Rocket Core "WithNSmallCores" fails to execute "rdcycle" instruction.

hahaha142142, 2022-06-12:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 
While I'm using Gemmini with Rocket Core "WithNSmallCores", I discover “rdcycle” instruction fails. Thus, I create a simple case to only test “rdcycle” instruction:
`#include <stdint.h>`
`int main() {`
`    uint64_t cycles;`
`    asm volatile ("rdcycle %0" : "=r" (cycles));`
`    return 0;`
`}`
The corresponding assembly code:
`0000000080001732 <main>:`
`    80001732:	c00027f3          	rdcycle	a5`
`    80001736:	4501                	li	a0,0`
`    80001738:	8082                	ret`
The execution result:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.`
`Listening on port 46373`
`[UART] UART0 is here (stdin/stdout).`
`*** FAILED *** (tohost = 1337)`
`*** FAILED *** (code = 1337, seed 1655149257) after 12587 cycles`
Then I annotate the "rdcycle" instruction:
`    //asm volatile ("rdcycle %0" : "=r" (cycles));`
The corresponding assembly code:
`0000000080001732 <main>:`
`    80001732:	4501                	li	a0,0`
`    80001734:	8082                	ret`
The execution result is different:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.`
`Listening on port 44037`
`[UART] UART0 is here (stdin/stdout).`
I think the question is closely related to Rocket Core "WithNSmallCores" rather than Gemmini. In fact, I'm not sure if it is a bug. Maybe I miss something.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Using Chipyard with Rocket Core "WithNSmallCores" and writing a simple case about "rdcycle" instruction.

**What is the current behavior?**
Execution fails.

**What is the expected behavior?**
Execution succeeds.

**Please tell us about your environment:**

Chipyard 1.6.2 with Rocket Core "WithNSmallCores".

**What is the use case for changing the behavior?**
The case about "rdcycle" instruction.

hahaha142142, 2022-09-15:
So, is this a bug?
I think "rdcycle" instruction intrinsically reads the "cycle" CSR of U-mode. Though I use the instruction in M-mode, it should be legal to read the "cycle" CSR of U-mode. Thus, the execution is supposed to succeed rather than fail.

ZenithalHourlyRate, 2022-09-15:
It is intended. See https://forums.sifive.com/t/read-csrs-rdcycle-on-rv32im/624/3

Instead you should use `asm volatile ("csrr %0, mcycle" : "=r" (cycles));`

hahaha142142, 2022-09-15:
> It is intended. See https://forums.sifive.com/t/read-csrs-rdcycle-on-rv32im/624/2
> 
> Instead you should use `asm volatile ("csrr %0, mcycle" : "=r" (cycles));`

Oh, I see. The "mcycle" CSR does work. Thank you!

However, I still feel something confused.
I know if there exists only M-mode, execution of "rdcycle" instruction in M-mode will fail due to lack of U-mode. Yet, if there exist both of M-mode and U-mode, I think "rdcycle" instruction is supposed to work in M-mode because CSRs in the lower privilege levels are accessible at all higher privilege levels (according to Volume 2, Privileged Spec v. 20211203).

In order to support my thinking, I run "rdcycle" instruction in M-mode on Spike, which is successful. Besides, I use "rv64ui-p-add" of isa case in riscv-test to confirm that Rocket Core "WithNSmallCores" indeed has U-mode.

So, if Rocket Core "WithNSmallCores" has both of M-mode and U-mode, why I cannot read the "cycle" CSR of U-mode by "rdcycle" instruction in M-mode?

ZenithalHourlyRate, 2022-09-15:
> I use "rv64ui-p-add" of isa cases in riscv-tests to confirm that Rocket Core "WithNSmallCores" indeed has U-mode.

This does not indicate that "WithNSmallCores" has U-mode. The test only uses "unprivileged" instructions, and it does not indicate which mode it is in. To check whether small core supports U-mode, you should check the u-bit in `misa`.

Actually, You can see that `WithNSmallCores` makes `useVM = false`, which leads to `hasSupervisorMode = false`, then `usingSupervisor = false` then `usingUser = false`. So there is no U-mode in this config.

hahaha142142, 2022-09-15:
> This does not indicate that "WithNSmallCores" has U-mode. The test only uses "unprivileged" instructions, and it does not indicate which mode it is in. To check whether small core supports U-mode, [...]

Got it, that's right. I have checked.
Thanks for answer!

closed 2022-09-15 by hahaha142142