chipsalliance/rocket-chip #3008: Rocket chip's AXI bus signals such as ready and RID seem to have some problems

qiuhaojie, 2022-07-11:
The file rocket-chip-master\src\main\scala\amba\axi4\Xbar.scala describes the AXI Slave interface, which we use to implement the PS to read and write registers to the PL through the AXI interface. Currently tested on bare metal, the write operation test works fine and the read operation test causes the PS to die. Using the Xil_Out32 () function in Vivado SDK for data writing, and function Xil_In32 () for data reading, found that only normal write once, can not write continuously, and can not read the data or the system will crash, we analyze is that some of the AXI bus signals are inconsistent with the normal AXI bus signals, such as the ARREADY signal default always high, and burst read and write RID signal and ARID signal also do not match, this will not be the cause of the system crash?

anjianfeng, 2022-07-11:
[image]

michael-etzkorn, 2022-07-12:
More explanation of your flow might help to pin down your problem. I doubt there's an issue with the AXI xbar generator. The zynq flow itself is deprecated as it hasn't been maintained for ~4 years https://github.com/ucb-bar/fpga-zynq