chipsalliance/rocket-chip #3018: compile fails

pxs7, 2022-07-31:
command:   mill -i rocketchip
error: 35/137] rocketchip.macros.compile
[info] compiling 1 Scala source to rocket-chip\out\rocketchip\macros\compile.dest\classes ...
[info] done compiling
[66/137] foreign-modules.hardfloat.hardfloat.compile
[info] compiling 20 Scala sources to  rocket-chip\out\foreign-modules\hardfloat\hardfloat\compile.dest\classes ...
[error] ## Exception when compiling 20 sources to  rocket-chip\out\foreign-modules\hardfloat\hardfloat\compile.dest\classes 
[error] scala.reflect.internal.Types$TypeError: type AffectsChiselPrefix is not a member of package chisel3.experimental
[error]
[error]
1 targets failed
hardfloat.compile scala.reflect.internal.Types$TypeError: type AffectsChiselPrefix is not a member of package chisel3.experimenta

michael-etzkorn, 2022-07-31:
It appears hardfloat's mill build is pointing to `3.5-SNAPSHOT` (which probably isn't a good idea). We'll have to have those point to stable versions so that mill doesn't cause this sort of eviction issue. 

https://github.com/ucb-bar/berkeley-hardfloat/blob/master/build.sc#L8-L9

pxs7, 2022-09-09:
> It appears hardfloat's mill build is pointing to `3.5-SNAPSHOT` (which probably isn't a good idea). We'll have to have those point to stable versions so that mill doesn't cause this sort of eviction issue.
> 
> If you changed these lines in hardfloat's build.sc to point to 3.5.3, that should fix the build issue.
> 
> https://github.com/ucb-bar/berkeley-hardfloat/blob/master/build.sc#L8-L9

some config has be set wrong，  and now all be ok

closed 2022-09-09 by pxs7