chipsalliance/rocket-chip #3019: icache and dcache with ECC (secded) enabled only works with zero initialized RAM arrays

tiwdi, 2022-08-01:
**Type of issue**: bug report | question

**Impact**: unknown

Applying ECC to to icache and dcache tag and data arrays works in simulation only if the RAM arrays are initialized with zero data.
While booting from APB attached ROM **it seems** that before writing program data to the icache, a read operation of random data causes a multibit error in the icache. It is just my guess, I have no proof that this is acutally the case. This operation is repeated infinitely.
The same simulation with zero initialized ichache works as expected. In a later stage, when dcache  is used, a similar problem occures.

```
   dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 128,
        nWays = 1,
        nTLBSets = 1,
        nTLBWays = 4,
        nMSHRs = 0,
        tagECC = Some("secded"),
        dataECC = Some("secded"),
        blockBytes = site(CacheBlockBytes))),
   icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 128,
        nWays = 1,
        nTLBSets = 1,
        nTLBWays = 4,
        tagECC = Some("secded"),
        dataECC = Some("secded"),
        blockBytes = site(CacheBlockBytes)))

```
Running the ECC cache rochet-chip on FPGA hardware (RAMs have random content after power up), the behavior seems to be like the simulation, whereas an implementation without ECC cache works fine. 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Im not sure that the problem is a bug, maybe I just implemented ECC the wrong way or missed some additional setting.

michael-etzkorn, 2022-09-15:
Random data would be inherently be equivalent to error data, and the proper data would need to be fetched from backing memory. I think this can likely be closed, and further discussion can happen on the discussion board if others are interested in looking at the case when RAM is initialized to non-zero.

tiwdi, 2022-09-15:
Yes, the expected behavior would be to fetch the data from backing memory but that does not happen. It simply hangs.
Zero data is also no useful code but it does not trigger the ECC to flag an non correctable error, so the data is read from the backing memory and property executed.
There seems to be a problem with error detection and cache hit/miss detection. Random data in tag RAM should most likely be a miss and therefore it does not matter if the cache data is valid or not.

BLackLmamba, 2024-02-14:
excuse me, how can I initialize the RAM arrays with zero data?and have you solved
 This problem?

sequencer, 2024-02-14:
For RAM that need to be initialized(metadata, etc) there are special logic to do it at reset. But RAM should be X. There is no way to initialize to any value by default. FPGA BRAM has such feature, but rocket doesn’t target to FPGA.

closed 2024-02-14 by sequencer