chipsalliance/rocket-chip #3032: CPU possibly hangs when a hella cache request is nacked

bgottschall, 2022-08-27:
**Type of issue**: bug report

**Impact**:  API modification

**Development Phase**: request

**Other information**
 

When the HellaCacheIO interface returns a request by pulling up s2_nack the SimpleHellaCacheIF replay queue will replay the request the next cycle. The HellaCacheArbiter will not let another pending HellaCacheIO request through. If the s2_nack can only be resolved by another entity using the hella cache interface the the CPU is hanging itself up.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

I've used a BOOM core with an RoCC implementation that issues a load request via the hella cache interface. This request will first go to the dTLB in the LSU and if it misses will return via the s2_nack signal. Now if this request has caused the dTLB to activate the page table walker, one will end up in a infinite loop because the PTW is also connected via the hella cache interface but the original request will block it by replaying over and over.

**What is the current behavior?**

Pipeline hangs

**What is the expected behavior?**

That the PTW can request through the hella cache interface while another request is in replay or simply by alternating requests in the arbiter.

**Please tell us about your environment:**

Using FireSim 1.14.0 with the BOOM core. I'm creating the bug issue here since all the classes that are effected are from the rocket-chip repository.

**What is the use case for changing the behavior?**

Resolving a possible problem that causes the CPU to hang.

jerryz123, 2022-08-27:
I believe this should be resolvable in BOOM. The arbiter between PTW and ROCC hellacache requests should prioritize the PTW.

bgottschall, 2022-08-27:
Yes that is of course the easiest solution I haven't thought of yet. Thanks!

closed 2022-08-27 by bgottschall

jerryz123, 2022-08-27:
Can you PR that fix to BOOM if it solves your problem?

bgottschall, 2022-08-28:
Yes will do it.