chipsalliance/rocket-chip #3070: Failed to enable the hypervisor extension correctly

bathtub-01, 2022-09-22:
I am currently trying to generate a build with hypervisor extension.
I have tried to enable it by setting a configuration in `rocket-chip/src/main/scala/system/Configs.scala`:
```Scala
class HypDefaultConfig extends Config(new WithHypervisor ++ new DefaultConfig)
```
Then I used
```
$ make verilog CONFIG=freechips.rocketchip.system.HypDefaultConfig
```
command under `rocket-chip/vsim` to build. The building process went fine, and I **did** find h-extension related CSRs in the generated Verilog file. But I only got `riscv,isa = "rv64imafdc"` in device tree. Seems like the hypervisor extension settings didn't fully applied.
Did I miss something when generating the chip?

ZenithalHourlyRate, 2022-09-22:
> Seems like the hypervisor extension settings didn't fully applied.

Well currently rocket chip just does not generate H in the DTS, see

https://github.com/chipsalliance/rocket-chip/blob/6e770de52c5e7bd6b664e9d617151818d4a9c92a/src/main/scala/tile/BaseTile.scala#L99-L109

bathtub-01, 2022-09-22:
Thank you. Does this mean that kvm cannot be applied out-of-the-box on the rocket chip?

ZenithalHourlyRate, 2022-09-22:
> Does this mean that kvm cannot be applied out-of-the-box on the rocket chip?

Yes. The linux kernel detects the "H" extension from DTS for now. You can modify this DTS or the process of generating this DTS to add the "H" character.

bathtub-01, 2022-09-23:
Thank you. That answers my question :)

closed 2022-09-23 by bathtub-01

michael-etzkorn, 2022-09-26:
Correct me if I'm wrong, but shouldn't we still fix this with a PR and leave the issue open?