chipsalliance/rocket-chip #3159: Assertion `endianness == memif_endianness_little || endianness == memif_endianness_big' failed

ncppd, 2022-10-25:
From https://github.com/riscv-software-src/riscv-isa-sim/issues/1129:

Hello, I am getting the error described in the title when I am trying to run tests in rocket chip. 

In order to get past this error and continue my work I hardcoded `virtual memif_endianness_t get_target_endianness()` to return `memif_endianness_little` in memif.h.

In order to build riscv-isa-sim I followed the instructions in the README. Have i configured something wrong, or is this a bug?

Assertion error:

>  emulator-freechips.rocketchip.system-freechips.rocketchip.system.DefaultConfig: ../fesvr/htif.h:35: T htif_t::from_target(target_endian<T>) const [with T = long unsigned int]: Assertion `endianness == memi      f_endianness_little || endianness == memif_endianness_big' failed.

Thanks!

jerryz123, 2022-10-25:
What command are you running in rocket-chip?

jerryz123, 2022-10-25:
Update to spike after https://github.com/riscv-software-src/riscv-isa-sim/pull/1130

closed 2022-10-25 by jerryz123