chipsalliance/rocket-chip #3165: Question regarding IO connection of `pma_checker` in `DCache`.

CircuitCoder, 2022-11-05:
**Type of issue**: (Potential?) bug report

**Impact**: unknown

**Development Phase**: request

**Other information**

In `rocket/DCache.sv`, `pma_checker` is connected as follows (https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/rocket/DCache.scala#L269):

```scala
pma_checker.io <> DontCare
pma_checker.io.req.bits.passthrough := true.B
pma_checker.io.req.bits <> s1_req
```

This connection seems to miss out most fields in TLBReq. Crucially, `vaddr` is not connected.

Currently the only usage of pma_checker's response is to set `s2_pma` when TLB is not ready, which is possible in some cases (e.g. if the type of memory operation is prefetch or fence). It seems to me that prefetches are not handled in DCache right now, and IIRC individual fences are not affected by PMAs? So the `pma_checker` code path is mostly redundant? If that's the case, can we remove them all together? If that's not the case, then this may be a bug.

This is found during the effort to convert rocket-chip to chisel3 (#3110), which changed the bi-directional connection semantic to disallowing missing fields in either side of the connection.

ZenithalHourlyRate, 2023-01-11:
> So the `pma_checker` code path is mostly redundant?

To fully understand the `pma_checker` we have to go back to AW's commit ["Expose DTLB access port"](https://github.com/chipsalliance/rocket-chip/pull/2270/commits/dc68c21e8e5662c81fba003a7579123735c757ef) which introduced this module.

In that commit, a `tlb_port` IO was added to `DCache`, and `DCache` has to handle requests from both `cpu` and `tlb_port`. In cases when we only have requests from one of them, requests are handled through `tlb` and everything is fine.

However, when we have requests from both of them, things become tricky. AW allowed one certain case: when the request from cpu is a physical address and has no exception. In all other cases, request from cpu is nacked.

https://github.com/chipsalliance/rocket-chip/blob/70ed8b63bd3780b2c6091a9e9fbfd12ab9145200/src/main/scala/rocket/DCache.scala#L265 

For that specific case, `pma_checker`, which is an extra TLB, only uses the PMPChecker part (that's why passthrough is always asserted true.B) and handles the request from cpu.

However, the author did not connect the request properly. `s1_req` is not of type `TLBReq` and has no `vaddr` field and should be manually connected in the following way instead of a bulk connection; but, Chisel2 allowed such connection and did not care about the unconnected port.

```chisel
  pma_checker.io.req.bits.vaddr := s1_req.addr
  pma_checker.io.req.bits.size := s1_req.size
  pma_checker.io.req.bits.cmd := s1_req.cmd
  // extra field from later commits
  pma_checker.io.req.bits.prv := s1_req.dprv
  pma_checker.io.req.bits.v := s1_req.dv
```

Fortunately, rocket-chip itself never used the extra `tlb_port` and is not affected. So the title should be changed to "PMA is not working in Rocket Core DCache when there are requests from cpu and tlb_port"

sequencer, 2023-01-11:
It spend me about 3 hours to think how to test this... And still have no idea on it.(Maybe eventually we need unittests to verify those uarch behavior???)

closed 2023-01-11 by sequencer