chipsalliance/rocket-chip #3207: Implementation of Zicntr

ZenithalHourlyRate, 2023-01-09:
**Type of issue**: feature request

**Impact**: API modification

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

For the zicntr extension, we have implemented the `cycle` and `instret` CSR (see https://github.com/chipsalliance/rocket-chip/commit/3ff921de046367dbd114150bfca471b5eb0c1550). However, to fully support zicntr, we still need to implement a `time` CSR.

This can not be simply done by connecting `mtime` to `time` as `mtime` is MMIO and currenly it is provided by clint.

**What is the expected behavior?**

There are two ways to provide the `time` CSR

* Hardwire `time` to `reg_cycle`. The spec said simple platform can do this; also, we have already used this fact by `csr.io.time`.
* Hardwire the RTC counter in [HasRTCModuleImp](https://github.com/chipsalliance/rocket-chip/blob/d706071fdc39233277a74d5332f566aa8358010e/src/main/scala/subsystem/RTC.scala#L24) to the CSR. However, not all possible config has the RTC module and CLINT.

The latter can be compliant with the spec on

> The real-time clocks of all harts must be synchronized to within one tick of the real-time clock.

Another final way is to provide `time` CSR from RTC if there is RTC; and fallback to `reg_cycle` if there is none. That will need to pass the `CLINTKey` to the `CSRFile`.

jerryz123, 2023-01-09:
I think just doing the simple solution is the best. Adding another clock crossing between the RTC and the tile would be additional complexity.

ZenithalHourlyRate, 2023-01-11:
Closed by #3208

ZenithalHourlyRate, 2023-01-12:
https://github.com/chipsalliance/rocket-chip/pull/3208#issuecomment-1379697646