chipsalliance/rocket-chip #3225: g++ version for `1.6` release ?

pcotret, 2023-01-14:
**Type of issue**: bug report 

**Impact**: API modification 

**Development Phase**: request 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
 I'm working with the `1.6` release. I was trying to compile the emulator (setup given below).
 - I had to change a flag in the emulator Makefile: https://github.com/chipsalliance/rocket-chip/pull/3194
 - Verilator 4.028 need a gnu++14 compatible compiler: https://github.com/verilator/verilator/blob/v4.028/configure.ac#L269

Is there a way to know which g++ version should be used?

**What is the expected behavior?**
Being able to compile the emulator.

**Please tell us about your environment:**
- Ubuntu 20.04.5 LTS
- g++ :
```bash
g++ -v       
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.1' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-Av3uEd/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 
```

michael-etzkorn, 2023-01-15:
The g++ version you linked doesn't seem to be problematic (that's what I use). Which version of riscv-tools are you on? 
v1.6 points to this hash: e2c6d1577a75f506fe992c3eb20a75174504476e. 

My fesvr doesn't include a cfg.h file? If I had a guess the version you're on is beyond what v1.6 supports. I use a toolchain built from chipyard v1.8 (which is compatible with RC v1.6).

michael-etzkorn, 2023-01-15:
That being said, this issue brings up a good point about documenting in the changelog the g++ and riscv-tools support.

pcotret, 2023-01-15:
...
It must be that, I was using the `master` branch without using the hash given in https://github.com/chipsalliance/rocket-chip/blob/1.6/riscv-tools.hash.

I'll try it later.

pcotret, 2023-01-15:
I had to update several submodules URL (my `git` didn't accept `git://<repo_url>` anymore) but it finally worked.

Here is my script to compile rocket-tools for the `v1.6` release: https://gist.github.com/pcotret/11afe52a1834172981c1e371b8bdcf03

michael-etzkorn, 2023-01-19:
Should be resolved by #3226, but we'll need to introduce a backport flow to fix v1.6 on checkout. I'll make a note about rocket-tools compatibility in the next changelog.

closed 2023-01-19 by michael-etzkorn