chipsalliance/rocket-chip #3228: Upgrade scripts to Python 3

Timmmm, 2023-01-16:
**Type of issue**: other enhancement
**Impact**: no functional change
**Development Phase**: proposal

There are a few Python scripts in this repo that use *Python 2*. This isn't installed by default in most repos anymore and I think it's even being removed from some. I don't really want to have my system unless some scripts accidentally use it - a lot of Python code (including these scripts!) just runs `python` which is sometimes Python 2 and sometimes Python 3, so you don't know which one it wants.

Fortunately the scripts seem to be quite small so it should be easy to update them.

See also #2629 - I also ran into this error and it's quite annoying because it doesn't say where the error originates from. Even worse, Ubuntu suggests installing `python-is-python3` which I'm guessing based on the name will just link `python` to `python3` and things will break even more.

Timmmm, 2023-01-16:
In fact even if you `apt-get install python2` it won't give you a `python` command. So at the very least you should change all the `#!/usr/bin/env python` lines to `python2`.

ZenithalHourlyRate, 2023-01-17:
Actually vlsi_mem_gen and vlsi_rom_gen are in python3 style and should suffice your use. I'll update the shebang to explicitly use python3.

As for trace script, I do not have test cases for them so I can not cover them.

jerryz123, 2023-01-17:
@ZenithalHourlyRate it would substantially improve test coverage to add the tracegen scripts to CI

sequencer, 2023-01-17:
FYI those script has been removed by 
https://github.com/chipsalliance/firrtl/pull/2202
and MFC has corresponding features, I think those scripts should be removed rather than bumping to py3

ZenithalHourlyRate, 2023-01-17:
> those script has been removed

At least vlsi_rom_gen is not implemented by firrtl.

I think we can bump to python3 so that users of old versions could cherry-pick that commit. Then we add the flow from firrtl and remove mem gen scripts.

The mill flow currently does not emit behav sram. It just outputs big reg arrays. That will be fixed later.