chipsalliance/rocket-chip #3230: Documentation generation failed

pcotret, 2023-01-18:
**Type of issue**: bug report 

**Impact**: unknown

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
```bash
git clone https://github.com/chipsalliance/rocket-chip
cd rocket-chip
git submodule update --init
sbt
```
In the SBT prompt:
```bash
docs/mdoc
```
Got following errors:
```
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] running mdoc.Main --cwd docs
[error] (run-main-0) java.lang.NoClassDefFoundError: scala/tools/nsc/reporters/AbstractReporter
[error] java.lang.NoClassDefFoundError: scala/tools/nsc/reporters/AbstractReporter
[error]         at java.base/java.lang.ClassLoader.defineClass1(Native Method)
[error]         at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1012)
[error]         at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:150)
[error] 	at java.base/java.net.URLClassLoader.defineClass(URLClassLoader.java:524)
[error] 	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:427)
[error] 	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:421)
[error] 	at java.base/java.security.AccessController.doPrivileged(AccessController.java:712)
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:420)
[error] 	at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:98)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:587)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:520)
[error] 	at mdoc.internal.markdown.MarkdownBuilder$.fromClasspath(MarkdownBuilder.scala:92)
[error] 	at mdoc.internal.cli.Context$.fromOptions(Context.scala:33)
[error] 	at mdoc.internal.cli.Settings.validate(Settings.scala:231)
[error] Caused by: java.lang.ClassNotFoundException: scala.tools.nsc.reporters.AbstractReporter
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:445)
[error] 	at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:98)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:587)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:520)
[error] 	at java.base/java.lang.ClassLoader.defineClass1(Native Method)
[error] 	at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1012)
[error] 	at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:150)
[error] 	at java.base/java.net.URLClassLoader.defineClass(URLClassLoader.java:524)
[error] 	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:427)
[error] 	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:421)
[error] 	at java.base/java.security.AccessController.doPrivileged(AccessController.java:712)
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:420)
[error] 	at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:98)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:587)
[error] Nonzero exit code: 1
[error] (docs / Compile / runMain) Nonzero exit code: 1
[error] Total time: 3 s, completed Jan 18, 2023, 10:59:44 AM
sbt:rocketchip> show discoveredMainClasses
[info] * freechips.rocketchip.groundtest.Generator
[info] * freechips.rocketchip.system.Generator
[info] * freechips.rocketchip.unittest.Generator

```

pcotret, 2023-01-18:
Made another try, my sbt/scala setup was a bit inaccurate.
Anyway, `docs/mdoc` still fails:
```bash
info: Compiling 5 files to ~/rocket-chip/docs/generated
error: ~/rocket-chip/docs/src/diplomacy/adder_tutorial.md:298:10: value io is not a member of freechips.rocketchip.diplomacy.LazyModuleImp
    when(monitor.module.io.error) {
         ^^^^^^^^^^^^^^^^^
info: Compiled in 2.92s (1 error)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (docs / Compile / runMain) Nonzero exit code: 1
[error] Total time: 102 s (01:42), completed Jan 18, 2023, 2:00:57 PM
```
Is there a pre-generated document file somewhere? (I'll read the source code in the meantime :wink: https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/rocket/ICache.scala#L19)

michael-etzkorn, 2023-01-19:
If you change that code block (and the two that follow) to `scala mdoc:fail`, you'll be able to compile ok, but the scala mdoc is bugged since monitor is being typed up to the parent class.