chipsalliance/rocket-chip #3255: Spurious data dependency in aes64ks1i instruction

tzwaenn, 2023-02-16:
**Type of issue**: bug report

**Impact**: performance impact

**Other information**
 
During formal verification with the BitManipCryptoConfig, we see the instruction in question being needlessly replayed. Consider the code sequence

        ld x19,256(x2)
        addi x8,x2,260
        aes64ks1i x11,x25,0x3

where the initial load has a cache miss. This is causing the aes64ks1i to replay without any actual dependency on the load. Apparently, the core thinks that aes64ks1i is using the rs2 field (encoding is actually 00 11000 1 rnum[3:0] rs1 001 rd 0010011), meaning a spurious dependency on x19 in this case with rnum=3.
I assume this is also causing needless stalls in decode.

This has been identified by running the Siemens Processor Verification App on the generated RTL.

ZenithalHourlyRate, 2023-02-16:
#3256 may fix the issue.

ZenithalHourlyRate, 2023-02-22:
Should be resolved now.

closed 2023-02-22 by ZenithalHourlyRate