chipsalliance/rocket-chip #3296: Problem with H (?) extension implementation?

gsomlo, 2023-03-13:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Building RocketChip:

```
cat >> rocket-chip/src/main/scala/subsystem/Configs.scala <<- "EOT"

	class WithMemoryDataBits(dataBits: Int) extends Config((site, here, up) => {
	  case MemoryBusKey => up(MemoryBusKey, site).copy(beatBytes = dataBits/8)
	})

	class WithLitexMemPort extends Config((site, here, up) => {
	  case ExtMem => Some(MemoryPortParams(MasterPortParams(
	                      base = x"8000_0000",
	                      size = x"8000_0000",
	                      beatBytes = site(MemoryBusKey).beatBytes,
	                      idBits = 4), 1))
	})

	class WithLitexMMIOPort extends Config((site, here, up) => {
	  case ExtBus => Some(MasterPortParams(
	                      base = x"1000_0000",
	                      size = x"7000_0000",
	                      beatBytes = site(SystemBusKey).beatBytes,
	                      idBits = 4))
	})

	class WithLitexSlavePort extends Config((site, here, up) => {
	  case ExtIn  => Some(SlavePortParams(
	                      beatBytes = site(SystemBusKey).beatBytes,
	                      idBits = 8,
	                      sourceBits = 4))
	})
	EOT

cat >> rocket-chip/src/main/scala/system/Configs.scala <<- "EOT"

	class BaseLitexConfig extends Config(
	  new WithLitexMemPort() ++
	  new WithLitexMMIOPort() ++
	  new WithLitexSlavePort ++
	  new WithNExtTopInterrupts(8) ++
	  new WithCoherentBusTopology ++
	  new BaseConfig
	)

	class FullLitexConfig extends Config(
	  new WithHypervisor ++
	  new WithBitManip ++ new WithBitManipCrypto ++
	  new WithCryptoNIST ++ new WithCryptoSM ++
	  new BaseLitexConfig
	)

	class LitexFull4DConfig extends Config(
	  new WithNBigCores(4) ++
	  new WithMemoryDataBits(128) ++
	  new FullLitexConfig
	)
	EOT

make RISCV=$HOME/RISCV -C rocket-chip/vsim verilog
```

The rocket chip as built above is then used with [LiteX](https://github.com/enjoy-digital/litex) to build a complete system capable of running Fedora (see https://github.com/litex-hub/linux-on-litex-rocket).

**What is the current behavior?**

Booting Fedora-configured kernel 6.3.0-rc1+ on 64-bit rocket with H support crashes during kernel kvm initialization
([here](https://github.com/torvalds/linux/blob/v6.3-rc2/arch/riscv/kvm/mmu.c#L765)), with following error:

```
...
[    7.048004] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    7.054576] NET: Registered PF_XDP protocol family
[    7.058534] PCI: CLS 0 bytes, default 64
[    7.079090] Oops - illegal instruction [#1]
[    7.082236] Modules linked in:
[    7.085262] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 6.3.0-rc1+ #14
[    7.091584] Hardware name: freechips,rocketchip-unknown (DT)
[    7.097222] epc : kvm_riscv_gstage_mode_detect+0x14/0x6a
[    7.102526]  ra : riscv_kvm_init+0x8e/0x142
[    7.106702] epc : ffffffff80c06d42 ra : ffffffff80c06bc2 sp : ffffffc80400bda0
[    7.113894]  gp : ffffffff82069238 tp : ffffffd8027d0000 t0 : ffffffc80400bd92
[    7.121092]  t1 : 036689f826b4fd31 t2 : 6f6e3d626c746f69 s0 : ffffffc80400bdb0
[    7.128306]  s1 : 0000000000000001 a0 : 0000000000000001 a1 : 0000000000000001
[    7.135512]  a2 : 0000000000000000 a3 : 0000000000000000 a4 : 0000000000000000
[    7.142724]  a5 : a000000000000000 a6 : 0000000000000003 a7 : 0000000000000010
[    7.149926]  s2 : ffffffff80c06b34 s3 : ffffffd802982000 s4 : 0000000000000000
[    7.157136]  s5 : ffffffff80e00da8 s6 : ffffffff80e00d60 s7 : ffffffff815ea7e8
[    7.164340]  s8 : ffffffff81527190 s9 : ffffffff80e92cb8 s10: 0000000000000008
[    7.171552]  s11: ffffffff80c000ea t3 : ffffffd81ddac808 t4 : 0000000000ff0000
[    7.178754]  t5 : 0000000200000000 t6 : ffffffff81fa62f8
[    7.184056] status: 0000000200000120 badaddr: 0000000068079073 cause: 0000000000000002
[    7.191978] [<ffffffff80c06d42>] kvm_riscv_gstage_mode_detect+0x14/0x6a
[    7.198582] [<ffffffff80c06bc2>] riscv_kvm_init+0x8e/0x142
[    7.204054] [<ffffffff8000275a>] do_one_initcall+0x5c/0x1c8
[    7.209608] [<ffffffff80c01854>] kernel_init_freeable+0x2a4/0x314
[    7.215692] [<ffffffff80bd7ffe>] kernel_init+0x2a/0x170
[    7.220896] [<ffffffff80003cda>] ret_from_exception+0x0/0x16
[    7.226788] Code: 0013 0000 0013 0000 1141 e022 e406 0800 57f5 17f6 (9073) 6807 
[    7.235718] ---[ end trace 0000000000000000 ]---
[    7.240556] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    7.247158] SMP: stopping secondary CPUs
[    7.251474] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```
**What is the expected behavior?**

kernel kvm initialization succeeds w/o kernel crashing

**Please tell us about your environment:**

 - version: `b2fd9918f` (git `dev` branch as of 03/13/2023)
 - OS: `Linux litex 6.3.0-rc1+ SMP Mon Mar 13 09:20:39 EDT 2023 riscv64 GNU/Linux`

**What is the use case for changing the behavior?**

Proper support for `H` extension. The generated sample `dts` contains `riscv,isa = "rv64imafdcZicsr_Zifencei_Zihpm_Xrocket";`, which, once linux boots, translates into `rv64imafdckph` in `/proc/cpuinfo`.

If I boot the kernel with just `riscv,isa = "rv64imafdc";` (i.e., I tell it no extensions are supported), it boots fine, because it won't try to actually use `H` related opcodes...

ZenithalHourlyRate, 2023-03-13:
> The generated sample `dts` contains `riscv,isa = "rv64imafdcZicsr_Zifencei_Zihpm_Xrocket"`

Ah, I know the problem now...The pitfall of CDE everyone will experience.

I now noticed that there is no `h` or `Zk*` or `Zb*` in your `riscv,isa`, which ought to be there.

That's because your order of applying config is wrong.

You should

```
	  new WithHypervisor ++
	  new WithBitManip ++ new WithBitManipCrypto ++
	  new WithCryptoNIST ++ new WithCryptoSM ++
	  new WithNBigCores(4) ++
```

Instead of

```
	  new WithNBigCores(4) ++
	  new WithHypervisor ++
	  new WithBitManip ++ new WithBitManipCrypto ++
	  new WithCryptoNIST ++ new WithCryptoSM ++
```

gsomlo, 2023-03-14:
@ZenithalHourlyRate -- thanks for pointing that out: adding `WithHypervisor`, `WithBitManip*`, and `WithCrypto*` ***before*** `WithNBigCores(*)` seems to result in a properly behaving set of Rocket cores. The isa string is now `rv64imafdchZicsr_Zifencei_Zihpm_Zba_Zbb_Zbc_Zbkb_Zbkc_Zbkx_Zbs_Zknd_Zkne_Zknh_Zksed_Zksh_Xrocket` (which Linux displays as `rv64imafdcbkph` in `/proc/cpuinfo`), and booting the kernel no longer crashes (during kvm initialization or otherwise).

A side effect is that a *properly* configured H-enabled Rocket system is significantly larger than before, but that's just to be expected :)

closed 2023-03-14 by gsomlo

ZenithalHourlyRate, 2023-03-14:
> adding `WithHypervisor`, `WithBitManip*`, and `WithCrypto*` _**before**_ `WithNBigCores(*)`

Just a comment: The config is evaluated in the following order 

```
(WithHypervisor ++ (WithBitManip ++ (WithCrypto ++ WithNBigCores(4))))
```

So strictly speaking it is not _**before**_, it is _**after**_

See also https://github.com/chipsalliance/cde/blob/384c06b8d45c8184ca2f3fba2f8e78f79d2c1b51/cde/src/chipsalliance/rocketchip/config.scala#L118-L127