chipsalliance/rocket-chip #3298: java.util.NoSuchElementException: RISCV

SpriteOvO, 2023-03-14:
**Type of issue**: bug report

After I ran `mill __.compile` under the project, I encountered error

```
Exception in thread "MillServerActionRunner" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:119)
        at java.base/java.lang.reflect.Method.invoke(Method.java:578)
        at mill.define.Module$Internal.$anonfun$reflect$4(Module.scala:70)
        at scala.collection.ArrayOps$WithFilter.map(ArrayOps.scala:90)
        at mill.define.Module$Internal.reflect(Module.scala:62)
        at mill.define.Module$Internal.reflectAll(Module.scala:73)
        at mill.define.Module$Internal.reflectNestedObjects(Module.scala:81)
        at mill.define.Module.millModuleDirectChildren$lzycompute(Module.scala:25)
        at mill.define.Module.millModuleDirectChildren(Module.scala:25)
        at mill.define.Module$Internal.rec$1(Module.scala:40)
        at mill.define.Module$Internal.$anonfun$traverse$1(Module.scala:40)
        at scala.collection.StrictOptimizedIterableOps.flatMap(StrictOptimizedIterableOps.scala:118)
        at scala.collection.StrictOptimizedIterableOps.flatMap$(StrictOptimizedIterableOps.scala:105)
        at scala.collection.immutable.ArraySeq.flatMap(ArraySeq.scala:35)
        at mill.define.Module$Internal.rec$1(Module.scala:40)
        at mill.define.Module$Internal.traverse(Module.scala:41)
        at mill.define.Module$Internal.modules$lzycompute(Module.scala:44)
        at mill.define.Module$Internal.modules(Module.scala:44)
        at mill.main.Resolve.resolve(Resolve.scala:211)
        at mill.main.RunScript$.$anonfun$resolveTasks$7(RunScript.scala:249)
        at scala.util.Either.map(Either.scala:382)
        at mill.main.RunScript$.$anonfun$resolveTasks$6(RunScript.scala:233)
        at scala.collection.immutable.List.map(List.scala:246)
        at scala.collection.immutable.List.map(List.scala:79)
        at mill.main.RunScript$.$anonfun$resolveTasks$5(RunScript.scala:232)
        at scala.util.Either.flatMap(Either.scala:352)
        at mill.main.RunScript$.resolveTasks(RunScript.scala:229)
        at mill.main.RunScript$.$anonfun$resolveTasks$2(RunScript.scala:202)
        at scala.collection.immutable.List.map(List.scala:246)
        at scala.collection.immutable.List.map(List.scala:79)
        at mill.main.RunScript$.$anonfun$resolveTasks$1(RunScript.scala:201)
        at scala.util.Either.flatMap(Either.scala:352)
        at mill.main.RunScript$.resolveTasks(RunScript.scala:200)
        at mill.main.RunScript$.evaluateTasks(RunScript.scala:312)
        at mill.main.RunScript$.$anonfun$runScript$8(RunScript.scala:105)
        at ammonite.util.Res$Success.flatMap(Res.scala:62)
        at mill.main.RunScript$.runScript(RunScript.scala:104)
        at mill.main.MainRunner.$anonfun$runScript$1(MainRunner.scala:119)
        at mill.main.MainRunner.watchLoop2(MainRunner.scala:67)
        at mill.main.MainRunner.runScript(MainRunner.scala:92)
        at mill.MillMain$.main0(MillMain.scala:320)
        at mill.main.MillServerMain$.main0(MillServerMain.scala:79)
        at mill.main.Server.$anonfun$handleRun$1(MillServerMain.scala:184)
        at java.base/java.lang.Thread.run(Thread.java:1589)
Caused by: java.util.NoSuchElementException: RISCV
        at scala.sys.package$.$anonfun$env$1(package.scala:69)
        at scala.collection.immutable.Map$WithDefault.default(Map.scala:180)
        at scala.collection.MapOps.apply(Map.scala:176)
        at scala.collection.MapOps.apply$(Map.scala:175)
        at scala.collection.AbstractMap.apply(Map.scala:405)
        at ammonite.$file.build.envByNameOrRiscv(build.sc:64)
        at ammonite.$file.build.envByNameOrRiscv$(build.sc:61)
        at ammonite.$file.build$.envByNameOrRiscv(build.sc:50)
        at ammonite.$file.build$riscv$minustests$.testsRoot(build.sc:239)
        at ammonite.$file.build$riscv$minustests$suite$.<init>(build.sc:246)
        at ammonite.$file.build$riscv$minustests$.suite$lzycompute$1(build.sc:245)
        at ammonite.$file.build$riscv$minustests$.suite(build.sc:245)
        at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104)
        ... 43 more
```

**Environment:**
    - version: 8b52a6f657823f1cfad703df8376d8f90c042ac0
    - OS: Arch Linux x86_64

sequencer, 2023-03-14:
https://github.com/chipsalliance/rocket-chip/blob/8b52a6f657823f1cfad703df8376d8f90c042ac0/build.sc#L64
I think we should blame this lane, for users don't need run CI environments, they should still be able to edit the codes.

ZenithalHourlyRate, 2023-03-14:
> blame this line

This is for rocket-tools.

For users that do not use nix, they should set `RISCV` from rocket-tools.

Or if they are not using rocket-tools, they should set `SPIKE_ROOT` and `RISCV_TEST_ROOT`.

>  edit the codes.

They are already exported in env var.

DavidZyy, 2024-03-12:
It seems that i get the same error in my metals' log:

```shell
2024.03.12 21:49:14 INFO  running '/home/zhuyangyang/project/ysyx-workbench/ysyxSoC/rocket-chip/.metals/millw --mill-version 0.11.7 --import ivy:com.lihaoyi::mill-contrib-bloop: mill.contrib.bloop.Bloop/install'[0m
2024.03.12 21:49:18 INFO  [info] compiling 4 Scala sources to /home/zhuyangyang/project/ysyx-workbench/ysyxSoC/rocket-chip/out/mill-build/compile.dest/classes ...[0m
2024.03.12 21:49:30 INFO  [warn] /home/zhuyangyang/project/ysyx-workbench/ysyxSoC/rocket-chip/build.sc:206:155: method any2stringadd in object Predef is deprecated (since 2.13.0): Implicit injection of + is deprecated. Convert to String to call +[0m
2024.03.12 21:49:30 INFO  [warn]          |"$${CMAKE_CXX_FLAGS} -DVERILATOR -DTEST_HARNESS=VTestHarness -include VTestHarness.h -include verilator.h -include ${generator.elaborate().path / config + ".plusArgs"}")[0m
2024.03.12 21:49:30 INFO  [warn]                                                                                                                                                           ^[0m
2024.03.12 21:49:30 INFO  [warn] one warning found[0m
2024.03.12 21:49:30 INFO  [info] done compiling[0m
2024.03.12 21:49:31 INFO  java.lang.reflect.InvocationTargetException[0m
2024.03.12 21:49:31 INFO      java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)[0m
2024.03.12 21:49:31 INFO      java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)[0m
2024.03.12 21:49:31 INFO      java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)[0m
2024.03.12 21:49:31 INFO      java.base/java.lang.reflect.Method.invoke(Method.java:568)[0m
2024.03.12 21:49:31 INFO      mill.define.Module$Internal.$anonfun$reflectNestedObjects$1(Module.scala:91)[0m
2024.03.12 21:49:31 INFO      scala.collection.immutable.ArraySeq.map(ArraySeq.scala:75)[0m
2024.03.12 21:49:31 INFO      scala.collection.immutable.ArraySeq.map(ArraySeq.scala:35)[0m
2024.03.12 21:49:31 INFO      mill.define.Module$Internal.reflectNestedObjects(Module.scala:90)[0m
2024.03.12 21:49:31 INFO      mill.define.Module.mill$define$Module$$millModuleDirectChildrenImpl(Module.scala:30)[0m
2024.03.12 21:49:31 INFO      mill.define.Module.mill$define$Module$$millModuleDirectChildrenImpl$(Module.scala:29)[0m
2024.03.12 21:49:31 INFO      millbuild.build$riscv$minustests$.mill$define$Module$$millModuleDirectChildrenImpl$lzycompute(build.sc:430)[0m
2024.03.12 21:49:31 INFO      millbuild.build$riscv$minustests$.mill$define$Module$$millModuleDirectChildrenImpl(build.sc:430)[0m
2024.03.12 21:49:31 INFO      mill.define.Module.millModuleDirectChildren(Module.scala:25)[0m
2024.03.12 21:49:31 INFO      mill.define.Module.millModuleDirectChildren$(Module.scala:25)[0m
2024.03.12 21:49:31 INFO      millbuild.build$riscv$minustests$.millModuleDirectChildren(build.sc:430)[0m
2024.03.12 21:49:31 INFO      mill.scalalib.internal.JavaModuleUtils$.loop$1(JavaModuleUtils.scala:17)[0m
2024.03.12 21:49:31 INFO      mill.scalalib.internal.JavaModuleUtils$.$anonfun$transitiveModules$1(JavaModuleUtils.scala:21)[0m
2024.03.12 21:49:31 INFO      scala.collection.immutable.ArraySeq.foldLeft(ArraySeq.scala:222)[0m
2024.03.12 21:49:31 INFO      mill.scalalib.internal.JavaModuleUtils$.loop$1(JavaModuleUtils.scala:21)[0m
2024.03.12 21:49:31 INFO      mill.scalalib.internal.JavaModuleUtils$.transitiveModules(JavaModuleUtils.scala:25)[0m
2024.03.12 21:49:31 INFO      mill.contrib.bloop.BloopImpl.computeModules(BloopImpl.scala:121)[0m
2024.03.12 21:49:31 INFO      mill.contrib.bloop.BloopImpl.install(BloopImpl.scala:30)[0m
2024.03.12 21:49:31 INFO      mill.contrib.bloop.BloopImpl.$anonfun$millDiscover$2(BloopImpl.scala:436)[0m
2024.03.12 21:49:31 INFO      mainargs.Invoker$.$anonfun$invoke0$2(Invoker.scala:77)[0m
2024.03.12 21:49:31 INFO      mainargs.Result.flatMap(Result.scala:13)[0m
2024.03.12 21:49:31 INFO      mainargs.Result.flatMap$(Result.scala:12)[0m
2024.03.12 21:49:31 INFO      mainargs.Result$Success.flatMap(Result.scala:23)[0m
2024.03.12 21:49:31 INFO      mainargs.Invoker$.invoke0(Invoker.scala:76)[0m
2024.03.12 21:49:31 INFO      mainargs.Invoker$.invoke(Invoker.scala:86)[0m
2024.03.12 21:49:31 INFO      mill.resolve.Resolve$.$anonfun$invokeCommand0$12(Resolve.scala:159)[0m
2024.03.12 21:49:31 INFO      mainargs.Result.flatMap(Result.scala:13)[0m
2024.03.12 21:49:31 INFO      mainargs.Result.flatMap$(Result.scala:12)[0m
2024.03.12 21:49:31 INFO      mainargs.Result$Success.flatMap(Result.scala:23)[0m
2024.03.12 21:49:31 INFO  jbva.util.NoSuchElementException: RISCV[0m
2024.03.12 21:49:31 INFO      scala.sys.package$.$anonfun$env$1(package.scala:69)[0m
2024.03.12 21:49:31 INFO      scala.collection.immutable.Map$WithDefault.default(Map.scala:181)[0m
2024.03.12 21:49:31 INFO      scala.collection.MapOps.apply(Map.scala:176)[0m
2024.03.12 21:49:31 INFO      scala.collection.MapOps.apply$(Map.scala:175)[0m
2024.03.12 21:49:31 INFO      scala.collection.AbstractMap.apply(Map.scala:405)[0m
2024.03.12 21:49:31 INFO      millbuild.build.envByNameOrRiscv(build.sc:405)[0m
2024.03.12 21:49:31 INFO      millbuild.build$riscv$minustests$.testsRoot(build.sc:411)[0m
2024.03.12 21:49:31 INFO      millbuild.build$riscv$minustests$suite$.<init>(build.sc:418)[0m
[... cut ...]
2024.03.12 21:49:31 INFO      mainargs.Result.flatMap(Result.scala:13)[0m
2024.03.12 21:49:31 INFO      mainargs.Result.flatMap$(Result.scala:12)[0m
2024.03.12 21:49:31 INFO      mainargs.Result$Success.flatMap(Result.scala:23)[0m
2024.03.12 21:49:32 INFO  time: ran 'mill bloopInstall' in 17s[0m
2024.03.12 21:49:32 ERROR Mill command failed: /home/zhuyangyang/project/ysyx-workbench/ysyxSoC/rocket-chip/.metals/millw --mill-version 0.11.7 --import ivy:com.lihaoyi::mill-contrib-bloop: mill.contrib.bloop.Bloop/install[0m
```

DavidZyy, 2024-03-13:
I solve this problem by install [rocket-tools](https://github.com/chipsalliance/rocket-tools) and set the `$RISCV` environment variable, and before you compile the codes, maybe you should see [this isuue](https://github.com/chipsalliance/rocket-tools/issues/85) first, the compilation maybe takes a few of hours.

DavidZyy, 2024-03-13:
I think it should remind people in the README to set these env vars properly.