chipsalliance/rocket-chip #3346: WithDebugSBA not working

RaphaelKlink, 2023-05-08:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

When WithDebugSBA is used after WithJTAGDTM, the debug interface is not SBA but JTAG

**What is the current behavior?**

the debug interface is JTAG

**What is the expected behavior?**

the debug interface is SBA

FIX:
class WithDebugSBA extends Config ((site, here, up) => {
  case ExportDebug => up(ExportDebug, site).copy(protocols = Set(DMI))
  case DebugModuleKey => up(DebugModuleKey, site).map(_.copy(hasBusMaster = true))
})

Can not upload my fix here, as i am working with a private gitlab

ZenithalHourlyRate, 2023-05-08:
> WithDebugSBA

SBA means SystemBusAccessState. It has nothing to do with the DTM.

RaphaelKlink, 2023-05-09:
Hey,

thanks for your speedy reply.

The Problem is, when With JtagDTM the ExportDebug Parameter is set to JTAG Interface and it is not reversed for the DebugSBA Option to the DMI Interface.

Jtag is the default option and therefore part of my abstract config. For specific workloads the SBA Interface is needed and therefore i added it.
For every other option i could simply overwrite my config option with another option. 

"Configs are additive, can override each other, and can be composed of other configs (sometimes referred to as config fragments)" is written in the chipyard documentation.

So i was simply confused, when this was not the case.

When i removed the WithJTAGDTM option in my abstract config it worked.

The Renaming would make things clearer.

Greetings
Raphael

RaphaelKlink, 2023-05-10:
That you for that clarification.

So the correct was would be to have a WithDMIDTM class and than not change the DebugSBA class at all?

ZenithalHourlyRate, 2023-05-10:
> So the correct was would be to have a WithDMIDTM class and than not change the DebugSBA class at all?

Yes.

michael-etzkorn, 2023-05-23:
Closing since the question looks resolved.

closed 2023-05-23 by michael-etzkorn