chipsalliance/rocket-chip #3377: How to make TraceGen configurations and scripts work?

GTwhy, 2023-05-23:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 
I want to generate mem tarce using the script `tracegen.py`. But I can't make the emulator with CONFIG=TraceGenConfig successfully. How to make these configurations and scripts work again?

https://github.com/chipsalliance/rocket-chip/blob/master/scripts/tracegen.py
> This script generates a trace using the given emulator (built
> with CONFIG=TraceGenConfig).  It waits until all cores have
> completed trace generation before terminating the emulator.

make and error log:
```
➜  emulator git:(master) ✗ make CONFIG=freechips.rocketchip.groundtest.TraceGenConfig                  
mkdir -p /home/hengyu/zspace/cpu/rocket-chip/emulator/generated-src/
cd /home/hengyu/zspace/cpu/rocket-chip && java -Xmx2G -Xss8M -cp /home/hengyu/zspace/cpu/rocket-chip/rocketchip.jar freechips.rocketchip.system.Generator -td /home/hengyu/zspace/cpu/rocket-chip/emulator/generated-src -T freechips.rocketchip.system.TestHarness -C freechips.rocketchip.groundtest.TraceGenConfig 
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at ... ()
        at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
        at ... ()
        at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined in Parameters
        at scala.Predef$.require(Predef.scala:337)
        at org.chipsalliance.cde.config$View.apply(config.scala:29)
        at freechips.rocketchip.subsystem.HasConfigurableTLNetworkTopology.$init$(BaseSubsystem.scala:62)
        at freechips.rocketchip.subsystem.BaseSubsystem.<init>(BaseSubsystem.scala:73)
        at freechips.rocketchip.subsystem.RocketSubsystem.<init>(RocketSubsystem.scala:33)
        at freechips.rocketchip.system.ExampleRocketSystem.<init>(ExampleRocketSystem.scala:11)
        at freechips.rocketchip.system.TestHarness.<init>(TestHarness.scala:16)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
        at chisel3.Module$.do_apply(Module.scala:53)
        at chisel3.stage.phases.Elaborate.$anonfun$transform$2(Elaborate.scala:40)
        at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:884)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
        at chisel3.internal.Builder$.build(Builder.scala:879)
        at chisel3.stage.phases.Elaborate.$anonfun$transform$1(Elaborate.scala:40)
        at scala.collection.immutable.List.flatMap(List.scala:293)
        at scala.collection.immutable.List.flatMap(List.scala:79)
        at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:28)
        at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:21)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
        at firrtl.options.DependencyManager.$anonfun$transform$5(DependencyManager.scala:280)
        at firrtl.Utils$.time(Utils.scala:181)
        at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:280)
        at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
        at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
        at scala.collection.immutable.List.foldLeft(List.scala:79)
        at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
        at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
        at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
        at chisel3.stage.ChiselStage.run(ChiselStage.scala:45)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[... 140 lines cut ...]
Such as:
```
make CONFIG=TraceGenConfig

get error:
java.lang.Exception: Unable to find part "TraceGenConfig" from "List(TraceGenConfig)", did you misspell it or specify the wrong package path?
```

```
make CONFIG=freechips.rocketchip.groundtest.TraceGenConfig
or
make CONFIG=freechips.rocketchip.groundtest.TraceGenConfig MODEL=TestHarness PROJECT=freechips.rocketchip.groundtest

get error:
java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocat
ed(Location(InSubsystem)) is not defined in Parameters 
```
**What is the expected behavior?**
Get an emulator with trace generator after make with TraceGenConfig.
**Please tell us about your environment:**
    - version: `47f7b7144727f0340d511d35b9f6c7a91b2a276f`
    - OS: `Linux 5.19.0-38-generic #39~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 21:16:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

**What is the use case for changing the behavior?**
Test memory, generate tarce and check by axe or other checkers.

ZenithalHourlyRate, 2023-05-23:
It is a known issue

https://github.com/chipsalliance/rocket-chip/issues/2920

GTwhy, 2023-05-23:
> It is a known issue
> 
> #2920

Thanks for your reply!
I have seen that issue, and I tried commenting out `IntSinkNode(IntSinkPortSimple()) :=* ibus.toPLIC`.
But I still got the error:
```
➜  emulator git:(master) ✗ make CONFIG=freechips.rocketchip.groundtest.TraceGenConfig PROJECT=freechips.rocketchip.groundtest
cd /home/hengyu/zspace/cpu/rocket-chip && java -Xmx2G -Xss8M -jar /home/hengyu/zspace/cpu/rocket-chip/sbt-launch.jar assembly
Warning: could not parse http_proxy setting: java.net.MalformedURLException: no protocol: 127.0.0.1:7890
Warning: could not parse https_proxy setting: java.net.MalformedURLException: no protocol: 127.0.0.1:7890
[info] welcome to sbt 1.3.13 (Private Build Java 1.8.0_362)
[info] loading settings for project rocket-chip-build from plugins.sbt ...
[info] loading project definition from /home/hengyu/zspace/cpu/rocket-chip/project
[info] loading settings for project rocketchip from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
Using addons: 
[info] set current project to rocketchip (in build file:/home/hengyu/zspace/cpu/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Strategy 'discard' was applied to 23 files (Run the task at debug level to see details)
[info] Strategy 'filterDistinctLines' was applied to a file (Run the task at debug level to see details)
[info] Assembly up to date: /home/hengyu/zspace/cpu/rocket-chip/rocketchip.jar
[success] Total time: 2 s, completed May 23, 2023 3:02:01 PM
mkdir -p /home/hengyu/zspace/cpu/rocket-chip/emulator/generated-src/
cd /home/hengyu/zspace/cpu/rocket-chip && java -Xmx2G -Xss8M -cp /home/hengyu/zspace/cpu/rocket-chip/rocketchip.jar freechips.rocketchip.groundtest.Generator -td /home/hengyu/zspace/cpu/rocket-chip/emulator/generated-src -T freechips.rocketchip.groundtest.TestHarness -C freechips.rocketchip.groundtest.TraceGenConfig 
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at ... ()
        at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
        at ... ()
        at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined in Parameters
        at scala.Predef$.require(Predef.scala:337)
        at org.chipsalliance.cde.config$View.apply(config.scala:29)
        at freechips.rocketchip.subsystem.HasConfigurableTLNetworkTopology.$init$(BaseSubsystem.scala:62)
        at freechips.rocketchip.subsystem.BaseSubsystem.<init>(BaseSubsystem.scala:73)
        at freechips.rocketchip.groundtest.GroundTestSubsystem.<init>(GroundTestSubsystem.scala:14)
        at freechips.rocketchip.groundtest.TestHarness.<init>(TestHarness.scala:12)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
        at chisel3.Module$.do_apply(Module.scala:53)
        at chisel3.stage.phases.Elaborate.$anonfun$transform$2(Elaborate.scala:40)
        at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:884)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
        at chisel3.internal.Builder$.build(Builder.scala:879)
        at chisel3.stage.phases.Elaborate.$anonfun$transform$1(Elaborate.scala:40)
        at scala.collection.immutable.List.flatMap(List.scala:293)
        at scala.collection.immutable.List.flatMap(List.scala:79)
        at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:28)
        at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:21)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
        at firrtl.options.DependencyManager.$anonfun$transform$5(DependencyManager.scala:280)
        at firrtl.Utils$.time(Utils.scala:181)
        at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:280)
        at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
        at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
        at scala.collection.immutable.List.foldLeft(List.scala:79)
        at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
        at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
        at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
        at chisel3.stage.ChiselStage.run(ChiselStage.scala:45)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[... 22 lines cut ...]
        at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
[... cut ...]
        at firrtl.options.Stage.transform(Stage.scala:47)
        at firrtl.options.Stage.execute(Stage.scala:58)
        at firrtl.options.StageMain.main(Stage.scala:71)
        at freechips.rocketchip.groundtest.Generator.main(Generator.scala)
make: *** [Makefrag-verilator:13: /home/hengyu/zspace/cpu/rocket-chip/emulator/generated-src/freechips.rocketchip.groundtest.TraceGenConfig.fir] Error 1
```
I looked at some issues and prs and tried a few things, but nothing worked, so I was a bit confused about what to do next.

ZenithalHourlyRate, 2023-05-23:
> I was a bit confused about what to do next.

I spent some time trying to solve this, and turned out in vain. The main reason might be that the Parameter, BaseTile and BaseSubsystem API and so on changed so much yet not much maintenance was given to ground test. It will take time to migrate to the new API.

Here is my debug process

> Caused by: java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined in Parameters

This was simple, just add specify one topology would suffice

```diff
 class TraceGenConfig extends Config(
   new WithTraceGen(2)() ++
+  new WithCoherentBusTopology ++
   new GroundTestBaseConfig
 )

```

The the interrupt connection is not good 

> Caused by: java.lang.NullPointerException: Cannot invoke "freechips.rocketchip.diplomacy.OutwardNodeHandle.outward()" because "h" is null                                                                    
>         at freechips.rocketchip.subsystem.CanAttachTile.connectInterrupts(HasTiles.scala:294)                                                                                                                

https://github.com/chipsalliance/rocket-chip/blob/0586532e47b679d472bed5dd2c70c246ba1da0c6/src/main/scala/subsystem/HasTiles.scala#L294

https://github.com/chipsalliance/rocket-chip/blob/0586532e47b679d472bed5dd2c70c246ba1da0c6/src/main/scala/groundtest/GroundTestSubsystem.scala#L30

https://github.com/chipsalliance/rocket-chip/blob/0586532e47b679d472bed5dd2c70c246ba1da0c6/src/main/scala/tile/Interrupts.scala#L31-L33

I assume this is connection issue as it further faults into the `:=` function

Tried #3322 and tried removing `:= domain { IntSyncAsyncCrossingSink(3) }`, not good. The later would result in

```
[#52] [error] src/main/scala/subsystem/HasTiles.scala:295:31: overloaded method := with alternatives:
[#52] [error]   [EY](h: freechips.rocketchip.diplomacy.OutwardNodeHandle[freechips.rocketchip.interrupts.IntSourcePortParameters,freechips.rocketchip.interrupts.IntSinkPortParameters,EY,chisel3.Vec[chisel3
.Bool]])(implicit p: org.chipsalliance.cde.config.Parameters, sourceInfo: chisel3.experimental.SourceInfo): freechips.rocketchip.diplomacy.NoHandle <and>
[#52] [error]   [DX, UX, EX, BX <: Chisel.Data, EY](h: freechips.rocketchip.diplomacy.NodeHandle[DX,UX,EX,BX,freechips.rocketchip.interrupts.IntSourcePortParameters,freechips.rocketchip.interrupts.IntSinkP
ortParameters,EY,chisel3.Vec[chisel3.Bool]])(implicit p: org.chipsalliance.cde.config.Parameters, sourceInfo: chisel3.experimental.SourceInfo): freechips.rocketchip.diplomacy.InwardNodeHandle[DX,UX,EX,BX]
[#52] [error]  cannot be applied to (freechips.rocketchip.interrupts.IntSyncOutwardNode)
[#52] [error]     domain.tile.intInwardNode := context.debugNode         
[#52] [error]                               ^                                                         
[#52] [error] one error found                                                                         

```

GTwhy, 2023-05-23:
Oh yes, I tried to add `WithCoherentBusTopology` and `WithIncoherentBusTopology` into `TraceGenConfig`, then I got those new errors.

I'm new to chisel and rocket-chip, trying to learn, and don't know what to do about these errors right now. I hope I can help with maintenance in the future.

Thank you for your time!