chipsalliance/rocket-chip #3383: Can the Makefile in emulator separate the firrtl and verilator, to allow modification on the generated verilog files?

fangrouli, 2023-06-08:
**Type of issue**: feature request

**Impact**: API modification | unknown

**Development Phase**: request

**Other information**
Hi, I am new to the Rocketchip structure, and am trying to learn through verilator simulation. So I really want my emulator testbench (`csrc/emulator.cc`) to be able to print out some internal signal value during execution of a binary file (e.g. the `ldut_mem_axi4_0_w_bits_data` that is connected to the `ExampleRocketSystem`. 

**What is the current behavior?**
While I try to search for how to do it on the Verilator platform, it advises me to add a `/*verilator public*/` after the declaration in the verilog file. But the issue is, every time I `make` in the emulator, the `.......DefaultConfig.v` will refresh based on the `Makefile` (retranslated from the scala file via firrtl), and gives a `error: ‘class VTestHarness’ has no member named ‘ldut’` error. So I am now really confused over how to peek into the dut.

I am also trying to figure out what are the individual commands that the `Makefile` is featuring, but it has a lot of referencing and it is hard for me to figure them out.

**What is the expected behavior?**
Is there a way to edit the scala file to make a certain signal viewable? Or maybe can we just separate the firrtl process from the `Makefile` of the emulator, so we can have a stable verilog file to work with at the testbench/binary execution stage?

**Please tell us about your environment:**
    - version: chisel 3.5.6,  verilator 4.226, Rocketchip 1.6
    - OS: Ubuntu 22.04.2 LTS
  
**What is the use case for changing the behavior?**
For peeking DUT's internal signal
If there is any other method can do this, could you please show me as well? It will greatly helps my learning! Thank you in advance!

ZenithalHourlyRate, 2023-06-08:
Hi, you might consider dumping vcd to get all the signal.

```bash
make debug # in emulator/ to get a debug emulator
./emulator-project-config-debug -v waveform.vcd pk helloworld
```

fangrouli, 2023-06-09:
> Hi, you might consider dumping vcd to get all the signal.

Yeah, I did that to view the signal. So is that the only way to check the internal signals in verilator simulation?

ZenithalHourlyRate, 2023-06-09:
> So is that the only way to check the internal signals in verilator simulation?

Well there are `printf`s in the chisel code, which will be generated as `$fwrite` in verilog with the condition PRINTF_COND. You might add `printf/frite` some where and modify the print cond in your emulator.cc or the generated verilog to print the signal in stdout. 

https://github.com/chipsalliance/rocket-chip/blob/0586532e47b679d472bed5dd2c70c246ba1da0c6/src/main/scala/rocket/RocketCore.scala#L1039-L1072

fangrouli, 2023-06-09:
Ohh I see it! 
I wrote this in TestHarness.scala
```printf ("Hello World in TestHarness\n")```

It will be converted to this in ....DefaultConfig.v
```
`ifndef SYNTHESIS
    `ifdef PRINTF_COND
      if (`PRINTF_COND) begin
    `endif
        if (_dtm_T_3) begin
          $fwrite(32'h80000002,"Hello World in TestHarness\n"); // @[TestHarness.scala 56:24]
        end
    `ifdef PRINTF_COND
      end
    `endif
`endif // SYNTHESIS
```
I think now I just need to figure out how to print the pin at the correct time. Thank you so much!

closed 2023-06-09 by fangrouli