chipsalliance/rocket-chip #3397: Removing Debug Module gives error

fangrouli, 2023-06-28:
**Type of issue**: bug report

**Impact**: no functional change

**Other information**
Repo: Rocket-chip 0586532e47b679d472bed5dd2c70c246ba1da0c6

System: Ubuntu 22.04

Purpose: Generate a DefaultRV32Config core without the debug module, because the .cc files leads to error when transferring the Verilog files to Centos6 terminal with VCS to compile. (As the VCS is 2014 and GNU is 4.7, no choice but to avoid any .cc files)
To execute programs I will try to read hex files into the ROM.

Error log:
```
/home/xxx/subsystem/RocketSubsystem.scala:33:7: class RocketSubsystem needs to be abstract.
[error] Missing implementation for member of trait DefaultTileContextType:
[error]   val debugNode: freechips.rocketchip.interrupts.IntSyncOutwardNode = ???
[error] class RocketSubsystem(implicit p: Parameters) extends BaseSubsystem with HasRocketTiles { //with HasPeripheryDebug {
[error]       ^
[warn] three warnings found
[error] one error found
[error] (Compile / compileIncremental) Compilation failed
```

**Steps to reproduce the problem:**
(following https://github.com/chipsalliance/rocket-chip/issues/1354 and https://github.com/chipsalliance/rocket-chip/issues/1046)
1. system/TestHarness.scala: 
(i) change 
`dut.reset := (reset.asBool | ldut.debug.map { debug => AsyncResetReg(debug.ndreset) }.getOrElse(false.B)).asBool` 
to `dut.reset := reset.asBool`
(ii) comment out `Debug.connectDebug(ldut.debug, ldut.resetctrl, ldut.psd, clock, reset.asBool, io.success)`

2. subsystem/RocketSubsystem.scala: **(The cause of error)**
(i) change 
`class RocketSubsystem(implicit p: Parameters) extends BaseSubsystem with HasRocketTiles with HasPeripheryDebug {` 
to `class RocketSubsystem(implicit p: Parameters) extends BaseSubsystem with HasRocketTiles {`

3. tile/Interrupts.scala:
(i) change `val async_ips = Seq(core.debug)`
to `core.debug := 0.U`
(ii) change `(async_ips ++ periph_ips ++ seip ++ core_ips).zip(interrupts).foreach { case(c, i) => c := i }`
to `(periph_ips ++ seip ++ core_ips).zip(interrupts).foreach { case(c, i) => c := i }`

**What is the current behavior?**
I think the error is due to me deleting the `HasPeripheryDebug` clashes the definition in `subsystem/HasTiles.scala`. But if I do not remove it, other errors complaining me not initializing all the debug pins will pop up. So is there anyway to workaround this and remove the Debug module? Can I do something like in [this](https://github.com/ucb-bar/chipyard/blob/ca9e132bc61950f4d5e06b191316097a1642d641/generators/chipyard/src/main/scala/config/fragments/PeripheralFragments.scala#L59) Chipyard configuration?

Thank you very much!

ZenithalHourlyRate, 2023-06-28:
Hi, you might try `case DebugModuleKey => None` instead of fiddle with the subsystem.

ZenithalHourlyRate, 2023-06-28:
> Hi, you might try `case DebugModuleKey => None` instead of fiddling with the subsystem.

fangrouli, 2023-06-28:
> Hi, you might try `case DebugModuleKey => None` instead of fiddling with the subsystem.

So I can just alter this into by config, without actually need to change anything in other files?!

ZenithalHourlyRate, 2023-06-28:
Yes. You might still need to alter the emulator.cc but generally no change is needed in scala sources.

fangrouli, 2023-06-28:
> Yes. You might still need to alter the emulator.cc but generally no change is needed in scala sources.

I add it into subsystem/Configs.scala with a new class, then quote this class in the system/Configs.scala, and it worked! 
But oddly if I directly add this into the Configs.scala with 
```
new BaseConfig()..alter((site,here,up) => {
    case DebugModuleKey => None})
```
like in the groundtest, it will produce this error:
```
[error] /home/xxx/system/Configs.scala:61:9: not found: value DebugModuleKey
[error] Identifiers that begin with uppercase are not pattern variables but match the value in scope.
[error]         case DebugModuleKey => None
[error] 
```
But anyways thank you so so so much for your fast help!

closed 2023-06-28 by fangrouli