chipsalliance/rocket-chip #3402: PTW incorrectly marks the access fault type for out-of-range non-leaf PTEs

poemonsense, 2023-06-29:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
In S mode, the PTW gets a valid, non-leaf PTE with a PPN exceeding the physical address space.

**What is the current behavior?**
Rocket-chip checks the PPNs (`!invalid_paddr` in PTW.scala:316) before traversing to the next level of page table. It thens returns `ae_final` to the TLB. However, since the page is non-leaf, it does not have the `r` bit set. Therefore, further a read request to TLB is responded with a page-fault exception.

**What is the expected behavior?**
According to the RISC-V manual 5.3.2, in step 4, if the PTE is valid and `pte.r = 0` and `pte.x = 0`, the PTE is a pointer to the next level of page table. Since we have not finished the three levels of page table, we continue to step 2 and `a = pte.ppn X PAGESIZE` to calculate the base address. If accessing the PTE violates the PMA, it should raise an access-fault exception.

This is the same behavior as in Spike. As seen in `mmu_t::walk`, if the PTE points to the next level of page table and does not have any bits of `PTE_D | PTE_A | PTE_U | PTE_N | PTE_PBMT` set, then we calculate the next base address with `base = ppn << PGSHIFT;` and continue to the next iteration of the for loop. In the next level, if accessing the PTE violates PMA (like exceeding the physical address space), an access-fault exception is raised.

**Please tell us about your environment:**

Found during co-simulating rocket-chip with riscv-isa-sim. Page-fault is delegated so rocket-chip goes to S mode while the Spike goes to M mode because of access-fault.
```
   mode different at pc = 0x0000011184, right= 0x0000000000000003, wrong = 0x0000000000000001
mstatus different at pc = 0x0000011184, right= 0x8000000a000060a2, wrong = 0x8000000a000060a8
sstatus different at pc = 0x0000011184, right= 0x8000000200006022, wrong = 0x8000000200006020
   mepc different at pc = 0x0000011184, right= 0x0000000000011186, wrong = 0x000000000001116e
   sepc different at pc = 0x0000011184, right= 0x0000000000010a1a, wrong = 0x0000000000011186
  mtval different at pc = 0x0000011184, right= 0x0000003f9bdf749e, wrong = 0x0000000000000000
  stval different at pc = 0x0000011184, right= 0x0000003f9c07ffac, wrong = 0x0000003f9bdf749e
 mcause different at pc = 0x0000011184, right= 0x0000000000000005, wrong = 0x0000000000000002
 scause different at pc = 0x0000011184, right= 0x000000000000000f, wrong = 0x000000000000000d
```

**What is the use case for changing the behavior?**
Just to align the behavior with Spike and the ISA. I don't think it would be a serious issue for users.

I notice the `pte_addr` is shortened after traversing to the next level. So I think it would be a little bit complicate to fix the behavior by allowing it traverse to the next level. Probably we should just set `ae_ptw` instead in this case.

poemonsense, 2023-06-29:
I tried a fix. This time rocket-chip raises the same access-fault as riscv-isa-sim does.

```
diff --git a/src/main/scala/rocket/PTW.scala b/src/main/scala/rocket/PTW.scala
index 7c36bdbf3..2e9fa8e7f 100644
--- a/src/main/scala/rocket/PTW.scala
+++ b/src/main/scala/rocket/PTW.scala
@@ -720,7 +720,11 @@ class PTW(n: Int)(implicit edge: TLEdgeOut, p: Parameters) extends CoreModule()(
           resp_valid(r_req_dest) := true.B
         }
 
-        resp_ae_final := ae
+        when (count <= 1.U) {
+          resp_ae_ptw := ae
+        }.otherwise {
+          resp_ae_final := ae
+        }
         resp_pf := pf && !stage2
         resp_gf := gf || (pf && stage2)
         resp_hr := !stage2 || (!pf && !gf && pte.ur())
```

Please correct me if I'm wrong. If the bug is confirmed, I can submit a PR to fix it.

poemonsense, 2023-06-30:
Probably I'm still missing the condition of reserved bits. Only if the non-leaf PTEs are valid and its PPN refers to an invalid address should we raise access-fault, because accessing the next-level PTE would violate the PMA.

Besides, I found that there may be a separate bug in the page-fault condition in PTW. Currently it checks only the most significant reserved bits (`val pf = pte.v && pte.reserved_for_future =/= 0.U`). However, as discussed in https://github.com/riscv-software-src/riscv-tests/issues/352, non-leaf should not have any of D, A, and U bits set. Therefore, rocket-chip needs to check the D, A, U bits as well for page-fault.

I believe rocket-chip's page-fault condition then would become exactly the same as Spike:
```
if (pte & (PTE_D | PTE_A | PTE_U | PTE_N | PTE_PBMT))
  break;
```

So there would be two separate bugs, for page-fault and access-fault respectively.

The missing cases are:
- access-fault: (1) it is a valid non-leaf PTE; (2) its PPN refers to an invalid address.
- page-fault: a non-leaf PTE has any of D, A, and U bits set.

poemonsense, 2023-06-30:
I've updated two commits to fix the two issues respectively.

- access-fault for non-leaf out-of-range PTE access: https://github.com/poemonsense/rocket-chip/commit/ff025a69dbc9b230de9a2ad386c9b333a92ea46c
```diff
diff --git a/src/main/scala/rocket/PTW.scala b/src/main/scala/rocket/PTW.scala
index 7c36bdbf3..8701caa0b 100644
--- a/src/main/scala/rocket/PTW.scala
+++ b/src/main/scala/rocket/PTW.scala
@@ -720,6 +720,7 @@ class PTW(n: Int)(implicit edge: TLEdgeOut, p: Parameters) extends CoreModule()(
           resp_valid(r_req_dest) := true.B
         }
 
+        resp_ae_ptw := ae && count < (pgLevels-1).U && pte.table()
         resp_ae_final := ae
         resp_pf := pf && !stage2
         resp_gf := gf || (pf && stage2)
```
For non-leaf PTEs (`pte.table()` is set, and `count` is less than the `pgLevels`), we should set `resp_ae_ptw` instead of `resp_ae_final`. Besides, since `resp_ae_final` would have lower priority than page-fault and `ae_ptw` access-fault, we don't need to shrink its condition and keep the condition as simple as possible. 

- page-fault for reserved, non-leaf PTES: https://github.com/poemonsense/rocket-chip/commit/c149a60d5dd71c471a7f846bccd2ae494278f1e9
```diff
diff --git a/src/main/scala/rocket/PTW.scala b/src/main/scala/rocket/PTW.scala
index 8701caa0b..be45281f7 100644
--- a/src/main/scala/rocket/PTW.scala
+++ b/src/main/scala/rocket/PTW.scala
@@ -135,8 +135,12 @@ class PTE(implicit p: Parameters) extends CoreBundle()(p) {
   val r = Bool()
   /** valid bit */
   val v = Bool()
+  /** the most significant bits are reserved for all types of PTE */
+  def isReserved: Bool = reserved_for_future =/= 0.U
+  /** a non-leaf PTE has more reserved bits than the leaf PTEs */
+  def nonLeafIsReserved: Bool = d || a || u
   /** return true if find a pointer to next level page table */
-  def table(dummy: Int = 0) = v && !r && !w && !x && !d && !a && !u && reserved_for_future === 0.U
+  def table(dummy: Int = 0) = v && !r && !w && !x && !isReserved && !nonLeafIsReserved
   /** return true if find a leaf PTE */
   def leaf(dummy: Int = 0) = v && (r || (x && !w)) && a
   /** user read */
@@ -695,7 +699,7 @@ class PTW(n: Int)(implicit edge: TLEdgeOut, p: Parameters) extends CoreModule()(
     }.otherwise {
       val gf = stage2 && !stage2_final && !pte.ur()
       val ae = pte.v && invalid_paddr
-      val pf = pte.v && pte.reserved_for_future =/= 0.U
+      val pf = pte.v && (pte.isReserved || count < (pgLevels-1).U && pte.nonLeafIsReserved)
       val success = pte.v && !ae && !pf && !gf
 
       when (do_both_stages && !stage2_final && success) {
```
I notice that, `table()` already has the correct checker logic for a valid non-leaf PTE in the bundle. Therefore, I just split the conditions and raise page-fault if any of the reserved bits for non-leaf PTEs is set.

Do you think these two bugs and fixes make sense? Notice that you added the valid `table()` condition for non-leaf PTEs. @aswaterman

ingallsj, 2023-07-11:
Thank you very much for bug reports, pull requests for fixes, and thorough descriptions; all excellent!  Much appreciated.

This will be closed by https://github.com/chipsalliance/rocket-chip/pull/3407.  For others, see that PR, and https://github.com/chipsalliance/rocket-chip/pull/3408, for discussion that we have already had on each.