chipsalliance/rocket-chip #3420: RVCExpander c.lui handling

chenguokai, 2023-07-13:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: proposal

**Other information**
 

According to the RISC-V spec, implements are free to expand a RVC HINT instruction into any other RVI HINT instruction that may not be of the same function. Though in rocket's implementation, other RVC instructions have been expanded into the same instruction in RVI form, whether it is a HINT or not. However, for `c.lui` which shares encoding space with `addi16sp`, there seems to be a disagreement with this implicit rule.

```scala
    def lui = {
      val opc = Mux(addiImm.orR, 0x37.U(7.W), 0x3F.U(7.W))
      val me = inst(Cat(luiImm(31,12), rd, opc), rd, rd, rs2p)
      Mux(rd === x0 || rd === sp, addi16sp, me)
    }
```

For `rd`=x0 case, it is expected that `c.lui` is expanded to a RVI `lui` while here `c.lui` is actually expanded to a RVI 'addi' which seems to reflect an `addi16sp`.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Pass a RVC `c.lui` instruction with `rd`=0 into the RVC expander, it will be expanded to `addi` rather than `lui`

**What is the current behavior?**

`c.lui` HINT is expanded to a RVI `addi` HINT.

**What is the expected behavior?**

`c.lui` HINT should be expanded to a RVI `lui` HINT.

**Please tell us about your environment:**

This bug was found in XiangShan(which has imported rocket as a depedency)'s verification flow.

**What is the use case for changing the behavior?**

For a unified logic for functional verification.

ZenithalHourlyRate, 2023-07-25:
Do not know if other cores are depending on this behavior. Might do a similar thing like https://github.com/chipsalliance/rocket-chip/commit/21722ddd51957f43b933f910f34b9c645fa54226

chenguokai, 2023-07-25:
Yes, I wonder if a similar PR is welcome :-)

ZenithalHourlyRate, 2023-07-25:
> Yes, I wonder if a similar PR is welcome :-)

I am OK. Other maintainers might complain about adding infinite options and things will become not maintainable.