chipsalliance/rocket-chip #3482: Calling `sb` instruction on code region memory does not change the memory content

trambu, 2023-09-01:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 
When we call `sb` instruction on a code memory address, Rocket core does not change the content or does not throw an exception. Note that no PMP  enabled and based on device tree and memory map configs the region is writeable. Also, on the Spike golden model memory content change. For more information, I attached device tree, memory map config, trace, spike trace, and sequence of instruction to reproduce.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Execute following instructions on Rocket core:
```
    8000104e:	00000297          	auipc	t0,0x0
    80001052:	00f28293          	addi	        t0,t0,15 # 8000105d <main+0x15>
    80001056:	00528023          	sb	        t0,0(t0)
    8000105a:	0001                	nop
    8000105c:	0001                	nop
    8000105e:	0001                	nop
```

**What is the current behavior?**
Rocket core does not change the content or does not throw an exception.

**What is the expected behavior?**
Rocket core has to change the content or throw an exception.

**Please tell us about your environment:**

    - version: `RocketChip v1.6`
    - OS: `CentOS Linux release 7.9.2009 kernel: 5.15.0-78-generic`
    - Simulator: `VCS_2021.09-SP1`
    - Spike version: `1.1.1-dev`
**What is the use case for changing the behavior?**
[trace_core.txt](https://github.com/chipsalliance/rocket-chip/files/12498476/trace_core.txt)
[memmap.txt](https://github.com/chipsalliance/rocket-chip/files/12498479/memmap.txt)
[dts.txt](https://github.com/chipsalliance/rocket-chip/files/12498484/dts.txt)
[spike_trace.txt](https://github.com/chipsalliance/rocket-chip/files/12498507/spike_trace.txt)

jerryz123, 2023-09-01:
How are you checking if the memory was actually changed?

trambu, 2023-09-01:
> How are you checking if the memory was actually changed?

As you see in the instruction sequence and spike log, I tried to change the content for the following instructions at `0x8000105d`. So, then when `pc` reaches that address, `0x8000105c`, the value has to be changed, but as you see in the rocket trace, the value is the same as before,`0x00000001,`, even though the spike trace shows that the value changed to `0x00005d01 `.

jerryz123, 2023-09-01:
FENCE.I is needed. RISC-V does not guarantee that stores are immediately visible to instruction fetch unless a FENCE.I is performed.

closed 2023-09-01 by jerryz123