chipsalliance/rocket-chip #3484: Rocket-chip emulator does not build /Make

sramichetty20019, 2023-09-16:
Hi all ,

I am new to rocket-chip, i have cloned the repository followed all the instructions to download and compile all the dependencies like riscv-tools,chisel,firrtl,verilator,yosys,expresso when i cd into the emulator directory and do make i get the following error .Another thing to mention is when i use sbt test with chisel some of the tests fail, same with firrtl..

this is the error i am getting 

~/rocket-chip/emulator$ make
cd /home/sramitchetty9/rocket-chip && mill rocketchip.assembly
/bin/bash: line 1: mill: command not found
make: *** [/home/sramitchetty9/rocket-chip/Makefrag:47: /home/sramitchetty9/rocket-chip/out/rocketchip/assembly.dest/out.jar] Error 127

please provide me with some information on how to resolve this

n-kremeris, 2023-10-03:
Hi!
You need to install "mill", which is a tool for building scala code. See info here: [mill installation](https://mill-build.com/mill/Installation_IDE_Support.html)

sramichetty20019, 2023-11-10:
Thank you , its working now

closed 2023-11-10 by sramichetty20019