chipsalliance/rocket-chip #3486: Error compiling `build.sc`

qianfei11, 2023-09-18:
**Type of issue**: bug report

**Impact**: unknown

~**Development Phase**: request | proposal~

~**Other information**~
 

**What is the current behavior?**

```bash
$ cd emulator/
$ make
cd /home/***/RISC-V/rocket-chip && mill rocketchip.assembly
Compiling /home/***/RISC-V/rocket-chip/build.sc
build.sc:594: not found: value Cross
trait JTAGDTMTest extends Cross.Module5[String, String, String, String, String] {
                          ^
build.sc:468: not found: value Cross
trait ArchTest extends Cross.Module4[String, String, String, String] {
                       ^
build.sc:440: not found: value Cross
trait RiscvTest extends Cross.Module4[String, String, String, String] {
                        ^
build.sc:113: not found: value Cross
trait Emulator extends Cross.Module2[String, String] {
                       ^
build.sc:68: not found: value millbuild
  extends millbuild.common.RocketChipModule
          ^
build.sc:71: not found: value Cross
    with Cross.Module[String] {
         ^
build.sc:56: not found: value millbuild
  extends millbuild.cde.common.CDEModule
          ^
build.sc:36: not found: value millbuild
  extends millbuild.hardfloat.common.HardfloatModule
          ^
build.sc:38: not found: value Cross
    with Cross.Module[String] {
         ^
build.sc:24: not found: value millbuild
  extends millbuild.common.MacrosModule
          ^
build.sc:426: not found: value Cross
  trait Suite extends Cross.Module[String] {
                      ^
build.sc:433: T{} members must be defs defined in a Cacher class/trait/object body
    def binaries = T {
                     ^
build.sc:429: T{} members must be defs defined in a Cacher class/trait/object body
    def description = T {
                        ^
build.sc:252: object Jvm is not a member of package mill.util
      mill.util.Jvm.runSubprocess(Seq("cmake", "-G", "Ninja", "-S", cmakefileLists().path, "-B", T.dest.toString).map(_.toString), Map[String, String](), T.dest)
                ^
build.sc:120: object Jvm is not a member of package mill.util
        mill.util.Jvm.javaExe,
                  ^
build.sc:253: object Jvm is not a member of package mill.util
      mill.util.Jvm.runSubprocess(Seq("ninja", "-C", T.dest).map(_.toString), Map[String, String](), T.dest)
                ^
build.sc:474: T{} members must be defs defined in a Cacher class/trait/object body
  def ispecString = T {
                      ^
build.sc:486: T{} members must be defs defined in a Cacher class/trait/object body
  def pspecString = T {
                      ^
build.sc:502: T{} members must be defs defined in a Cacher class/trait/object body
  def spikeRoot = T {
                    ^
build.sc:506: T{} members must be defs defined in a Cacher class/trait/object body
  def CC = T {
             ^
build.sc:86: not found: value crossValue
  def hardfloatModule = hardfloat(crossValue)
                                  ^
build.sc:82: not found: value crossValue
  def chiselPluginIvy = Some(v.chiselCrossVersions(crossValue)._2)
                                                   ^
build.sc:80: not found: value crossValue
  def chiselIvy = Some(v.chiselCrossVersions(crossValue)._1)
                                             ^
build.sc:50: not found: value crossValue
  def chiselPluginIvy = Some(v.chiselCrossVersions(crossValue)._2)
                                                   ^
build.sc:48: not found: value crossValue
  def chiselIvy = Some(v.chiselCrossVersions(crossValue)._1)
                                             ^
build.sc:48: `T.command` definitions must have 1 parameter list
  def chiselIvy = Some(v.chiselCrossVersions(crossValue)._1)
      ^
build.sc:114: not found: value crossValue
  val top: String = crossValue
                    ^
build.sc:115: not found: value crossValue2
  val config: String = crossValue2
                       ^
build.sc:427: not found: value crossValue
    val name: String = crossValue
                       ^
build.sc:441: not found: value crossValue
  val top: String = crossValue
                    ^
build.sc:442: not found: value crossValue2
  val config: String = crossValue2
                       ^
build.sc:443: not found: value crossValue3
  val suiteName: String = crossValue3
                          ^
build.sc:444: not found: value crossValue4
  val exclude: String = crossValue4
                        ^
build.sc:469: not found: value crossValue
  val top: String = crossValue
                    ^
build.sc:470: not found: value crossValue2
  val config: String = crossValue2
                       ^
build.sc:471: not found: value crossValue3
  val xlen: String = crossValue3
                     ^
build.sc:472: not found: value crossValue4
  val isa: String = crossValue4
                    ^
build.sc:595: not found: value crossValue
  val top: String = crossValue
                    ^
build.sc:596: not found: value crossValue2
  val config: String = crossValue2
                       ^
build.sc:597: not found: value crossValue3
  val sba: String = crossValue3
                    ^
build.sc:598: not found: value crossValue4
  val xlen: String = crossValue4
                     ^
build.sc:599: not found: value crossValue5
  val name: String = crossValue5
                     ^
Compilation Failed
make: *** [/home/***/RISC-V/rocket-chip/Makefrag:47: /home/***/RISC-V/rocket-chip/out/rocketchip/assembly.dest/out.jar] Error 1
```

~**What is the expected behavior?**~

**Please tell us about your environment:**

- version: 50adbdb3e
- OS: `Linux bea1e-OptiPlex-5000 5.15.0-83-generic #92~20.04.1-Ubuntu SMP Mon Aug 21 14:00:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

~**What is the use case for changing the behavior?**~

sequencer, 2023-09-18:
Please use nix, it seems that you are using the old version of mill, which
is not supported.

qianfei11, 2023-09-18:
> Please use nix, it seems that you are using the old version of mill, which is not supported.
> […](#)

Fix the error after i upgrade mill to 0.11.3. After this, it tells me `Cannot resolve rocketchip.assembly.`:

```bash
$ make
cd /home/bea1e/RISC-V/rocket-chip && mill rocketchip.assembly
[build.sc] [48/52] compile
[info] compiling 4 Scala sources to /home/bea1e/RISC-V/rocket-chip/out/mill-build/compile.dest/classes ...
[warn] /home/bea1e/RISC-V/rocket-chip/build.sc:206:155: method any2stringadd in object Predef is deprecated (since 2.13.0): Implicit injection of + is deprecated. Convert to String to call +
[warn]          |"$${CMAKE_CXX_FLAGS} -DVERILATOR -DTEST_HARNESS=VTestHarness -include VTestHarness.h -include verilator.h -include ${generator.elaborate().path / config + ".plusArgs"}")
[warn]                                                                                                                                                           ^
[warn] one warning found
[info] done compiling
[build.sc] [52/52] methodCodeHashSignatures
Cannot resolve rocketchip.assembly. Try `mill resolve rocketchip._` or `mill resolve __.assembly` to see what's available.
make: *** [/home/bea1e/RISC-V/rocket-chip/Makefrag:47: /home/bea1e/RISC-V/rocket-chip/out/rocketchip/assembly.dest/out.jar] Error 1
```

Here is the result of executing `mill resolve` in the root directory of rocket-chip: 

```bash
$ mill resolve rocketchip._
[1/1] resolve
rocketchip[3.6.0]
rocketchip[5.0.0]
$ mill resolve __.assembly
[1/1] resolve
cde.assembly
hardfloat[3.6.0].assembly
hardfloat[5.0.0].assembly
macros.assembly
rocketchip[3.6.0].assembly
rocketchip[3.6.0].cdeModule.assembly
rocketchip[3.6.0].hardfloatModule.assembly
rocketchip[3.6.0].macrosModule.assembly
rocketchip[5.0.0].assembly
rocketchip[5.0.0].cdeModule.assembly
rocketchip[5.0.0].hardfloatModule.assembly
rocketchip[5.0.0].macrosModule.assembly
```