chipsalliance/rocket-chip #3495: make failing under /emulator and /vcs due to mill: command not found

AnuragMalwee, 2023-09-22:
### Discussed in https://github.com/chipsalliance/rocket-chip/discussions/3492

<div type='discussions-op-text'>

<sup>Originally posted by **AnuragMalwee** September 20, 2023</sup>
Hi, I am building the rocket-chip project for the first time.

After building the rocket-tools, the make command under $ROCKETCHIP/emulator and $ROCKETCHIP/vcs directories fail with error:
```
      cd ...../Rocket/rocket-chip && mill rocketchip.assembly
      /bin/bash: line 1: mill: command not found
      make: *** [/home/...../Rocket/rocket-chip/Makefrag:47: /home/...../Rocket/rocket-chip/out/rocketchip/assembly.dest/out.jar] Error 127
```

When trying to install mill from https://github.com/com-lihaoyi/mill, the cmd to run the bootstrap script recommended in their documentation fails.
$ curl -L https://github.com/com-lihaoyi/mill/releases/download/0.11.4/0.11.4-13-27c979 > mill && chmod +x mill
(The mill/releases dir doesn't seem to exist in the repo)

I manually copied the mill/mill script and ran it in the $ROCKETCHIP directory but I got following error:

```
anurag@anurag-vm:~/...../Rocket/rocket-chip$ ~/...../scala_mill/mill 
[build.sc] [33/49] enablePluginScalacOptions.super.mill.scalalib.ScalaModule.enablePluginScalacOptions | Downloading [1/1][build.sc] [33/49] enablePluginScalacOptions.super.mill.scalalib.ScalaModule.enablePluginScalacOptions | Downloading [1/1][build.sc] [33/49] enablePluginScalacOptions.super.mill.scalalib.ScalaModule.enablePluginScalacOptions | Downloading [1/1][build.sc] [33/49] enablePluginScalacOptions.super.mill.scalalib.ScalaModule.enablePluginScalacOptions | Downloading [1/1][build.sc] [41/49] compile 
[info] compiling 4 Scala sources to /...../Rocket/rocket-chip/out/mill-build/compile.dest/classes ...
[warn] /...../Rocket/rocket-chip/build.sc:206:155: method any2stringadd in object Predef is deprecated (since 2.13.0): Implicit injection of + is deprecated. Convert to String to call +
[warn]          |"$${CMAKE_CXX_FLAGS} -DVERILATOR -DTEST_HARNESS=VTestHarness -include VTestHarness.h -include verilator.h -include ${generator.elaborate().path / config + ".plusArgs"}")
[warn]                                                                                                                                                           ^
[warn] one warning found
[info] done compiling
[build.sc] [49/49] scriptImportGraph 
Selector cannot be empty
```

</div>

SpinEch0, 2023-12-20:
add mill into your exec path: export PATH="/path/to/app/executable/file/directory:$PATH"

jackkoenig, 2023-12-20:
A lot of projects commit `millw` [[1]](https://mill-build.com/mill/Installation_IDE_Support.html#_millw) (as `mill`) to the root of the repository to simplify installation (eg. scala-cli does this, although they have both an older `mill` bootstrap script and `millw`). I suggest rocket-chip does the same @sequencer @jerryrzhao.