chipsalliance/rocket-chip #3508: firtool: Unknown command line argument '-disable-infer-rw'.  Try: 'firtool --help'

zzulb, 2023-10-11:
firtool: Unknown command line argument '-disable-infer-rw'.  Try: 'firtool --help'

This is my first installation. When I execute "make verilog" directly, it prompts " firtool: Unknown command line argument '-disable-infer-rw'.  Try: 'firtool --help' " error. The new version of firtool does not have the "-disable-infer-rw" option. I am using 1.38.0.

sequencer, 2023-10-11:
Please use nix version of Circt this issues has been fixed by #3506

closed 2023-10-11 by sequencer

sequencer, 2023-10-11:
I just merged #3506, you can have a try with dev branch, but please use circt from nix in the future since we only support one circt in rc

youzi27, 2023-10-11:
Title: Error using Nix version of Circt with RocketChip Emulator

Description:
I recently switched to the Nix version of Circt as suggested, but encountered a new error when compiling with RocketChip Emulator.

Error:
```
1 targets failed
emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile os.SubprocessException: Result of firtool…: 1

os.proc.call(ProcessOps.scala:89)
millbuild.build$Emulator$mfccompiler$.$anonfun$compile$2(build.sc:152)
mill.define.Task$TraverseCtx.evaluate(Task.scala:71)
```
I appreciate any assistance or suggestions to resolve this issue. Thank you!

icanflyo, 2023-10-23:
> Title: Error using Nix version of Circt with RocketChip Emulator
>
> Description: I recently switched to the Nix version of Circt as suggested, but encountered a new error when compiling with [...]

Hello, I have also encountered the same problem. Have you resolved this issue?

sequencer, 2023-10-23:
please use nix.

wangzhiwei18, 2023-11-22:
> please use nix.

Please elaborate on how to build circt from Nix.
Sorry, I'm a rookie.

youzi27, 2023-11-22:
> Hello, I have also encountered the same problem. Have you resolved this issue?

Yes, I solved this problem by using the Nix package manager. You should enable the Nix functionality that comes with the repository. For example: `nix-shell ....` 
Please refer to the knowledge on how to use Nix features.