chipsalliance/rocket-chip #3522: How to generate rocket chip in verilog

arunvthampi, 2023-11-06:
I am a beginner in using rocket chip. I tried to follow the commands. And I think I got an some system verilog files in ./out/emulator/compile.dest. What should I do to get the rocket chip in verilog. Do I need to change some lines of code in build.sc

Hoping to receive a reply and thank you

jerryz123, 2023-11-06:
Try 'make verilog' in the root of rocketchip

arunvthampi, 2023-11-07:
> Try 'make verilog' in the root of rocketchip

Yeah I followed all the commands including 'make verilog'. But I am getting the files in system verilog

sequencer, 2023-11-07:
there is no Verilog support anymore in the CIRCT project

arunvthampi, 2023-11-07:
Thank you for your reply

closed 2023-12-22 by arunvthampi

khoatm98, 2024-05-05:
Hi @sequencer , may I ask about generated files if we dont support verilog anymore? I tried to find documents and threads mentioning about this but haven't found. After running 'make verilog', I can only see *.class files, which I dont know whether it's correct or not? 
[image]

jerryz123, 2024-05-05:
Check Chipyard for a flow that generates rocket-chip-based SoCs with Verilog emission.