chipsalliance/rocket-chip #3524: How to generate Boom core  on other FPGA platforms like zcu102

ProJeKTRED, 2023-11-10:
As a beginner, I have searched for other projects that generate Boom cores, but they are no longer available due to updates to the boom core and git loss. I only have zcu102 in our lab. Thank u.
https://github.com/li3tuo4/rc-fpga-zcu

jerryz123, 2023-11-10:
Currently the supported mechanism is to add more FPGA board support to Chipyard. This question is best directed to the Chipyard repo, which supports BOOM. Rocket-chip is an independent project.

ProJeKTRED, 2023-11-10:
> Currently the supported mechanism is to add more FPGA board support to Chipyard. This question is best directed to the Chipyard repo, which supports BOOM. Rocket-chip is an independent project.

ok, thanks for you reply. So i should submit issue in the feature request of chipyard?

jerryz123, 2023-11-10:
Yes.

closed 2023-11-10 by jerryz123