chipsalliance/rocket-chip #3529: Remove Scalar Crypto

sequencer, 2023-11-15:
TL;DR
==

I'm going to remove the scalar crypto support:

Here are the detail reason:
- Due to decision in RISCV-Profiles:
> The scalar crypto extensions Zkn and Zks that were options in RVA22 are not options in RVA23. The goal is for both hardware and software vendors to move to use vector crypto, as vectors are now mandatory and vector crypto is substantially faster than scalar crypto. 
- reducing the maintaining overhead, the original contributor left and left RTL w/o documentation and support.
- the current implementation is not in a good architecture: it doesn't use the OoO WB feature in RocketChip, thus only has one cycle for execution; it has a bad design in uop, decoding stage.

In the future, we might add it back with a better architecture and comprehensive documentation and implementation.

jerryz123, 2023-11-15:
I agree with this reasoning.

jerryz123, 2023-11-15:
Speaking more broadly, I believe we should avoid committing RTL of new functional units into this repository, and focus on developing generator interfaces and APIs that allow for plugging things into Rocket.

sequencer, 2023-11-15:
I do think so, we might need more refactoring to deliver a comprehensive interface, rather than adding more implementation to the codebases.

jerryz123, 2023-11-15:
That's ok, its much easier to reason about the correctness of refactored generator code, than refactored logic

closed 2023-11-17 by sequencer